LazeMSS/OctoPrint-TopTemp #5: CPU Temp pathing

Pheelix, 2021-01-24:
Not sure if the command path to my  vcgencmd isn't default.  But I had to change the pathing in settings to /usr/bin/vcgencmd measure_temp.   Had to modify some navbartemp files as well with same path, but will be using toptemp from here on out.

If my pathing to vcgencmd isn't the normal path then what you have defaulted, then go ahead and close this post.  But this is the pathing it gave me from a default install of octoprint on a RPI4.

LazeMSS, 2021-01-24:
My plugin tests available know options and show the detected options as predefined. I think the RPI in the later distros have them in both paths. Basically if something is shown in the list under predefined means it worked when the plugin started.

Please post anything you have found that works for you/you have changed so I can test.

Pheelix, 2021-01-24:
I just modifited the command to: 
/usr/bin/vcgencmd measure_temp|cut -d "=" -f2|cut -d"'" -f1
and it works perfectly.  Just had to replace /opt/vc/bin/ for the command above.

On another note.  I changed the Icons for bed and tool:
Bed: fas fa-solar-panel
Tool: fas fa-filter
I was trying to find a Line with heat was above it for bed and something close to what looks like a hotend but didn't manage to find anything :(.   So these seem pretty close.

Also a few suggestions.  Could you space out the Icon and the Text a little or maybe add a option to add X amount of space.
Text Size and Color options would be nice too.  Tho I have already added Custom CSS code for the size and color.  Couldn't figure out how to space the icons and text.  My CSS coding skills suck.

I know this is a new plugin and you are still working on it.  Just my input on it.  Liking it so far.

[image]

Pheelix, 2021-01-24:
Some quick photoshoping of what I was thinking.

[image]
[image]

A option to put the Icons on the sides seems nice.  Again all up to you.

LazeMSS, 2021-01-24:
There are two options tested regarding vcgencmd:
/usr/bin/vcgencmd measure_temp|cut -d "=" -f2|cut -d"'" -f1
/opt/vc/bin/vcgencmd measure_temp|cut -d "=" -f2|cut -d"'" -f1

So the one you posted should be on the list and added - both off them normally work - well its the same command normaly just with an alias/symlink.

I dont allow colors on the icons and text at the moment - this is done by using standard boostrap styling - this is to avoid conflicts with any themes etc. Although i might add it later.

I will look into having the icons to the different sides and more spacing.

Pheelix, 2021-01-24:
Yep no problem on the colors, I can just use Custom CSS for that.

As for the vcgencmd.  This was the other predefined. cat /sys/class/thermal/thermal_zone0/temp | sed 's/\(.\)..$/.\1/'
I didn't see the predefined to start with as I was using the normal Tree List on the left instead of the UI Customizer version (mobile/responsive) that puts them at the top.   Didn't notice the scroll bar at the bottom so that is my fault.  Also as I am not that good with linux based alias/symlink, I just used the one I had to add to navbartemp to fix the SoC.

Anyways, looking forward to seeing more additions and improvements.  If you need me to test anything just let me know. :)

LazeMSS, 2021-01-24:
Awesome thanks for getting back - forgot to test without UI customizer. That is fixed in next version.

SureshotM6, 2021-01-24:
> As for the vcgencmd. This was the other predefined. cat /sys/class/thermal/thermal_zone0/temp | sed 's/\\(.\\)..$/.\1/'

Note this that command doesn't round up properly and also would have issues around 0^C (which really shouldn't be expected unless someone keeps their Pi in a freezer though).

I would suggest to use this command instead.  It's also less CPU time to run (although both are tiny):
`awk '{print $0/1000}' /sys/class/thermal/thermal_zone0/temp`

ken536, 2021-01-25:
I would like an option to ask for temperatures using the terminal (I use klipper)

LazeMSS, 2021-01-25:
> I would like an option to ask for temperatures using the terminal (I use klipper)

Im not sure how that works. I have no experience with kipper

ken536, 2021-01-25:
I fact real simple: I send TEMP_CASE to the terminal and it responds with a value I can set in the option. This could be a single number:

> Send: TEMP_CASE
Recv: // Case:
Recv: // Temperature: 37.05 C
Recv: // Pressure: 1002.53 hPa
Recv: // Humidity: 11.93%

This way I could also use it to show the humidity

LazeMSS, 2021-01-25:
@ken536  please post as seperate report :)

LazeMSS, 2021-02-02:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.0.3

I'm closing for now - if not fixed please reopen this :)

closed 2021-02-02 by LazeMSS