LazeMSS/OctoPrint-TopTemp #6: Line settings (width, color, and opacity) are not honored

SureshotM6, 2021-01-24:
If you change the graph line setting, the lines still show as 1px width, black colored, and 40% opacity.  Tested with Chrome 87.0.4280.141.

The default "#navbar_plugin_toptemp div.TopTempGraph .ct-line" CSS rule, defined in [TopTemp.css](https://github.com/LazeMSS/OctoPrint-TopTemp/blob/main/octoprint_toptemp/static/css/TopTemp.css#L77) is taking effect.

The javascript is generating dynamic CSS rules like this:
`#TopTempGraph_bed_graph{height:50%};#TopTempGraph_bed_graph.TopTempGraph > svg >g .ct-line{stroke-width: 2px; stroke-opacity: 0.2; stroke: #0000ff;}`

The first semicolon and the closing curly brace are swapped, so the rule does not apply.  Swapping these back like so fixes the issue:
`#TopTempGraph_bed_graph{height:50%;} #TopTempGraph_bed_graph.TopTempGraph > svg >g .ct-line{stroke-width: 2px; stroke-opacity: 0.2; stroke: #0000ff;}`

Relevant line of js code [here](https://github.com/LazeMSS/OctoPrint-TopTemp/blob/main/octoprint_toptemp/static/js/TopTemp.js#L892).

Thanks for the plugin!

LazeMSS, 2021-01-24:
Thanks - did alot of panic fixed before release :)

LazeMSS, 2021-01-25:
Should be fixed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.0.2

Please reopen if not fixed/working.

closed 2021-01-25 by LazeMSS