LazeMSS/OctoPrint-TopTemp #8: CPU temp "VCHI initialization failed"

GlassOnTin, 2021-01-24:
I'm running on [Raspberry Pi OS](https://www.raspberrypi.org/software/), with OctoPrint: 1.5.3.

The default "CPU temperature" command is:
```bash
/opt/vc/bin/vcgencmd measure_temp | cut -d "=" -f2 | cut -d"'" -f1
```
and when I press "Test" I see:
```
Error:   VCHI initialization failed.
```

Google searching the "VCHI initialization failed" error shows that this is a vcgencmd permissions problem, e.g.:
[Chewett.co.uk blog post](https://chewett.co.uk/blog/258/vchi-initialization-failed-raspberry-pi-fixed/)

Solution, I added the "octoprint" user to the "video" group AND restarted OctoPrint.  The error message was still there until I restarted OctoPrint.

LazeMSS, 2021-01-24:
Are you running octopi from an image installation or did you install it yourself?

LazeMSS, 2021-01-24:
And did you try any of the predefined options - I know they are hidden to the right if you are not running UI Customizer :(

LazeMSS, 2021-01-27:
Closing - feedback needed - @GlassOnTin reopen if you can answer my questions.

closed 2021-01-27 by LazeMSS