LazeMSS/OctoPrint-TopTemp #12: Readouts keep disapearing

TrotFoxGreyfoot, 2021-01-25:
Hi!

I love the concept but I've noticed that the feedback readouts keep disapearing every few seconds, resulting in the display length shortening to only the setpoints. I see a flag icon pop up when the readback disapears but don't see a reference to it in the docs so I'm not sure what it means? Mostly it's just slightly irritating but a stable display length might be something to look at.

Thanks,
Landon

Normal:
[image]

Lost T0/Shortened:
[image]

LazeMSS, 2021-01-25:
What version are you running?

LazeMSS, 2021-01-25:
It looks like the previous version - 0.0.0.1 - please try updating to 0.0.0.2

LazeMSS, 2021-01-25:
Are the target temperature is not show when the target is reached - instead of showing 200/200 it just displays 200 and a checkered flag to show its reached the target.
You can uncheck "Show target progress"

LazeMSS, 2021-01-25:
Closing - its by design - I might add an option to always show target temperature :)

closed 2021-01-25 by LazeMSS