LazeMSS/OctoPrint-TopTemp #9: Editing anything under customMon.*.graphSettings breaks plugin until server restart

SureshotM6, 2021-01-25:
Reproduced on both of my octoprint instances, but it took me a second to figure out what caused it and reproduce it again.

Procedure:
1. Start from a fresh plugin install (delete toptemp in config.yaml).  **Not required for issue** but easiest to reproduce this way.
2. Open toptemp settings and change anything under CPU Temperature Background Graph settings, such as "Graph transparency".
3. Save settings
4. Refresh the web browser
5. Note the periodic errors in the browser developer console, such as "Error calling fromCurrentData on view model TopTempViewModel : TypeError: settings.height is not a function" coming from TopTempViewModel.self.setGraphStyle.
6. Open toptemp settings again
7. Settings aren't correctly displayed.  Custom "CPU Temperature" monitor is missing.  New tabs such as "Chamber settings" and Extra tools Tool 1-9 are visible.
8. Developer console shows errors such as "Error calling onSettingsShown on view model TopTempViewModel : TypeError: curItem is not a function" coming from TopTempViewModel.self.buildCustomSettings.

If you look at config.yaml in this case, only the modified settings under plugins.toptemp.customMon.cu0.graphSettings are present.  The other settings are absent.

A restart of the server fixes the issue.  After the server restart, "Fixing custom monitors" is logged by the plugin and plugins.toptemp.customMon.cu0.graphSettings looks fully populated in config.yaml.  However, if I go back to step #2 above and again set the "CPU Temperature" "Graph transparency" to something else, the issue reproduces another time until a server restart.

I'm not too familiar with this octoprint configuration framework, but I would imagine that a quick fix may be to run this "Fixing custom monitors" / _merge_dictionaries code when the settings are being loaded.  I think you might be able to override on_settings_load() and perform the merge with default settings there instead of within on_settings_initialized()?

SureshotM6, 2021-01-25:
I just noticed that if I go back and edit "CPU Temperature" height, then my changes to opacity are lost in the config file.  If I then change "line width", then my modified opacity comes back in the config file, but then my height changes are lost.  There must be something else wrong here too.

Anyway, hope I'm not annoying you by opening too many issues.  Thanks for making the plugin again!

LazeMSS, 2021-01-25:
Thank you for a perfect and execellent bug report 👍

LazeMSS, 2021-01-25:
Should be fixed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.0.2

Please reopen if not fixed/working.

closed 2021-01-25 by LazeMSS