LazeMSS/OctoPrint-TopTemp #13: Add Hotend and Part Cooling Fan Speed

lightmaster, 2021-01-26:
I don't know how to do it, but it would be nice to be able to show the speeds of various fans using this. At the vary least, the Part Cooling Fan, but also the Hotend Fan if its set to be automatic instead of always on.

LazeMSS, 2021-01-26:
Normally these cant be read but I will look into getting them from the gcode sent

lightmaster, 2021-01-27:
Not sure how they do it, but DisplayLayerProgress is able to get the cooling part fan, just type in `[fanspeed]` where you want it:
```
Fan: <span style='display: inline-block;width:42px;'>[fanspeed]</span>
```

LazeMSS, 2021-01-27:
Yes I was guessing they were using the gcode - this will not work if you change the fan speed directly on the printer. But its better than nothing. I will look into it and add it together with the other stuff new stuff I will be adding.

LazeMSS, 2021-02-04:
@lightmaster  which gcode can enable/disable the hotend fan? I know of M106 for fanspeed but not hotend fand

lightmaster, 2021-02-04:
Hot end is either always on with 12/24v constant, or hooked up to a mosfet and automatically on when hotend temp is above a certain to set in `Configuration_adv.h`. If only for safety reasons, I doubt you can directly control the hotend fan with gcode.

LazeMSS, 2021-02-19:
This is fixed in the next release - please test/try by following the below guide:
```
Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install
```

lightmaster, 2021-02-20:
Once I noticed the default settings for `Only while printer is printing` and disabled that, looks like it's working great.

LazeMSS, 2021-02-20:
Reopen will close when public release is out 😄

LazeMSS, 2021-02-21:
Closing - fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.0

Please reopen if not fixed

closed 2021-02-21 by LazeMSS