LazeMSS/OctoPrint-TopTemp #15: Add Bluetooth Thermometer Monitoring

RChadwick7, 2021-01-26:
Those cheap $4 bluetooth thermometers are great for dry boxes, and when you install updated firmware they are very easy to read.

LazeMSS, 2021-01-26:
I don't have that hardware so it's kinda hard to make it but I'm guessing you could probably get the data using a script. Do you have a link for the hardware you are using?

RChadwick7, 2021-01-27:
This is a link that describes the open-source firmware. The device itself is available on Aliexpress or Ebay. I bought 3 for $12 including shipping. They are pretty accurate, and the new firmware was extremely easy to flash.

https://hackaday.com/2020/12/08/exploring-custom-firmware-on-xiaomi-thermometers/

LazeMSS, 2021-01-27:
Well basically you can just do it yourself - write that code below (from your link) and put it file called bttemp.py - and run it as a custom command with a grep command or you could make it easier and modify it like below - then it will only return the temperature.

`#!/usr/bin/env python3
import sys
from datetime import datetime
import bluetooth._bluetooth as bluez
 
from bluetooth_utils import (toggle_device, enable_le_scan,
                             parse_le_advertising_events,
                             disable_le_scan, raw_packet_to_str)
 
# Use 0 for hci0
dev_id = 0
toggle_device(dev_id, True)
 
try:
    sock = bluez.hci_open_dev(dev_id)
except:
    print("Cannot open bluetooth device %i" % dev_id)
    raise
 
# Set filter to "True" to see only one packet per device
enable_le_scan(sock, filter_duplicates=False)
 
try:
    def le_advertise_packet_handler(mac, adv_type, data, rssi):
        data_str = raw_packet_to_str(data)
        # Check for ATC preamble
        if data_str[6:10] == '1a18':
            temp = int(data_str[22:26], 16) / 10
            hum = int(data_str[26:28], 16)
            batt = int(data_str[28:30], 16)
            print("%sc" % \
                 (mac))
 
    # Called on new LE packet
    parse_le_advertising_events(sock,
                                handler=le_advertise_packet_handler,
                                debug=False)
# Scan until Ctrl-C
except KeyboardInterrupt:
    disable_le_scan(sock)`

Im closing this because adding in code hardware support for hardware that I dont own and can't easy buy is out of scope and there is an easy fix as highlighted above

closed 2021-01-27 by LazeMSS