LazeMSS/OctoPrint-TopTemp #16: Terminal temp command

ken536, 2021-01-26:
When using klipper a lot of information can be requested using the terminal.

Information from sensors can be requested in the terminal:

> Send: TEMP_CASE
Recv: // Case:
Recv: // Temperature: 37.05 C
Recv: // Pressure: 1002.53 hPa
Recv: // Humidity: 11.93%

Information can be changed in the klipper settings so it only returns a number.

LazeMSS, 2021-02-04:
Just to make sure - so you want to make an option where this Gcode command is sent at X interval and when this happens look for the reply. This could be a problem to implement in a good way and I dont think there is many users with a kipper setup like this.

I might add this down the line but for now I'm trying to make the basic stuff work

ken536, 2021-02-05:
It is in fact the same as '#18 by fdraeger' (M105 in terminal), but klipper gives you the option to name to mcodes to anything you want, in my case: 'TEMP_CASE'. 
So you can close this and if the solution for fdraeger is with a custum option for the mcode, this one is solved as well!

LazeMSS, 2021-02-05:
I will try.

LazeMSS, 2021-02-05:
The problem is have no way to test this - and don't you need to send a command to get this data?

ken536, 2021-02-05:
We also need to send the M105 on the terminal right?

LazeMSS, 2021-02-05:
No i think the temperatures are constantly sent back from the printers.

LazeMSS, 2021-02-19:
This should be fixed in the next release - please test/try by following the below guide:
```
Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install
```

LazeMSS, 2021-02-21:
Closing - fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.0

Please reopen if not fixed

ken536, 2021-02-22:
I've tried but I can't get it to work. I'm not sure what to insert in the field, but I've created this from the examples.

I've tried this. The return value is : '// 19.22'. I've tried the reg exp and this should work on this return value.

[image]

LazeMSS, 2021-02-22:
You need a pattern for each value - so for case temp:
"Type" = "From printer"
Regexp: `Temperature: ([^ ]+)[^<>\r\n]`

I think this might work.

ken536, 2021-02-22:
I've added an extra gcode calles TEMP_CASE_SIMPLE

> Send: TEMP_CASE_SIMPLE
> Recv: // 37.05

I need the tool to send 'TEMP_CASE_SIMPLE' and listen for the response. I should use 'GCode sent' isn't it?

LazeMSS, 2021-02-22:
My plugin does not support sending any commands. It only listens for outgoing gcode (thats gcode sent) and any data comming back. 

That was why original wrote (No the temperatures are constantly sent back from the printers.) - so if you need to send a command and then listen for the response then I current don't have support for that.

Can't klippe just expand the M105 response to include the other stuff like marlin and others does it:
Send: M105 Recv: ok T:18.3 /0.0 B:18.6 /0.0 T0:18.3 /0.0 @:0 B@:0 P:19.3 A:29.2 C:37
etc.

ken536, 2021-02-22:
Now I understand what you mean by that.
I will try both options later today!

ken536, 2021-02-22:
I can't get it to work with the original idea. Adding to M105 doesn't seem to work either:

[image]
[image]

I've tested the Reg Exp:
[image]

LazeMSS, 2021-02-22:
Can you show a screenshot where it is added to M105 in the terminal output?

ken536, 2021-02-23:
I wasn't able to add it to my M105 code, but I've tried to read T0 from the M105 response. (I've changed the code you gave as example in #18, and checked tested the reg exp which was fine). Reading this doesn't seem to work either

LazeMSS, 2021-02-23:
@ken536 so either the output of klipper is different from others using or something else i broken.
When you say you are trying to read the T0 what do you mean?

Try installing the develop version:
Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install

And see if that works.

ken536, 2021-02-23:
From the 

> Send: M105
> Recv: ok B:19.1 /0.0 T0:19.0 /0.0

I'll try the develop version in a minute

LazeMSS, 2021-02-23:
That is a complete broken M105 response to be honest.

LazeMSS, 2021-02-23:
But you should still be able to grap for whatever you like

ken536, 2021-02-23:
Why is the M105 broken?

[image]

I've tried both now:
> CT: ([^ ]+)[^<>\r\n]

>T0:([^ ]+)[^<>\r\n]

LazeMSS, 2021-02-23:
Normally the start with T0:

Nothing happens - and you are using the "From printer" type and using the develop I posted?

You can try and uncomment this line:

https://github.com/LazeMSS/OctoPrint-TopTemp/blob/develop/octoprint_toptemp/__init__.py#L764

And then tail the log files: tail -F  /home/pi/.octoprint/logs/octoprint.log

ken536, 2021-02-23:
This is my output after changing the code so an update of that is visible in the log as well:

> 2021-02-23 11:57:19,445 - octoprint.plugins.toptemp - INFO - New gcode gcIn mon needed: cu2:CT: ([^ ]+)[^<>\r\n]
2021-02-23 11:57:19,446 - octoprint.plugins.toptemp - INFO - Setting up custom gcode monitor for "cu2". Type: gcIn pattern: CT: ([^ ]+)[^<>\r\n]
2021-02-23 11:57:32,936 - octoprint.plugins.toptemp - INFO - cat /sys/class/thermal/thermal_zone0/temp | sed 's/\(.\)..$/.\1/' returned: 62.3 for index :cu0
2021-02-23 11:57:32,938 - octoprint.plugins.toptemp - INFO - Got custom data: 62.3
2021-02-23 11:57:32,940 - octoprint.plugins.toptemp - INFO - Sending data to UI, cu0 : 62.3

It only repeats this all the time

> 2021-02-23 11:56:17,777 - octoprint.plugins.toptemp - INFO - cat /sys/class/thermal/thermal_zone0/temp | sed 's/\(.\)..$/.\1/' returned: 63.7 for index :cu0
2021-02-23 11:56:17,780 - octoprint.plugins.toptemp - INFO - Got custom data: 63.7
2021-02-23 11:56:17,782 - octoprint.plugins.toptemp - INFO - Sending data to UI, cu0 : 63.7
2021-02-23 11:56:42,834 - octoprint.plugins.toptemp - INFO - cat /sys/class/thermal/thermal_zone0/temp | sed 's/\(.\)..$/.\1/' returned: 63.2 for index :cu0
2021-02-23 11:56:42,836 - octoprint.plugins.toptemp - INFO - Got custom data: 63.2
2021-02-23 11:56:42,838 - octoprint.plugins.toptemp - INFO - Sending data to UI, cu0 : 63.2
2021-02-23 11:57:07,886 - octoprint.plugins.toptemp - INFO - cat /sys/class/thermal/thermal_zone0/temp | sed 's/\(.\)..$/.\1/' returned: 60.3 for index :cu0
2021-02-23 11:57:07,890 - octoprint.plugins.toptemp - INFO - Got custom data: 60.3
2021-02-23 11:57:07,890 - octoprint.plugins.toptemp - INFO - Sending data to UI, cu0 : 60.3

But I did notice this error while loading the plugin:

> 2021-02-23 11:55:52,570 - octoprint.plugins.toptemp - INFO - Setting up custom gcode monitor for "cu3". Type: gcIn pattern: T0:([^ ]+)[^<>\r\n]
2021-02-23 11:55:52,572 - octoprint.plugins.toptemp - INFO - Setting up custom gcode monitor for "cu2". Type: gcIn pattern: CT: ([^ ]+)[^<>\r\n]
2021-02-23 11:55:52,585 - octoprint.plugins.toptemp - INFO - Setting up custom timer for "cat /sys/class/thermal/thermal_zone0/temp | sed 's/\(.\)..$/.\1/'(cu0 / cmd) running each 25 seconds
2021-02-23 11:55:52,606 - octoprint.plugin - ERROR - Error while calling plugin toptemp
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_toptemp/__init__.py", line 113, in on_after_startup
    threading.Thread(target=self.gcodeRecvQworker, daemon=True).start()
TypeError: __init__() got an unexpected keyword argument 'daemon'

LazeMSS, 2021-02-23:
Argh you are running python 2 🙈

I will check and see if there is a way to get it working on python 2.

LazeMSS, 2021-02-23:
I think I have fixed the problem with the daemon - try again:

```
Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install
```

ken536, 2021-02-24:
Thats it!

[image]

LazeMSS, 2021-02-24:
https://www.paypal.com/paypalme/ihccaptain

Feel free to donate :)

ken536, 2021-02-24:
[image]

LazeMSS, 2021-02-24:
Closing - is now released.

closed 2021-02-24 by LazeMSS