LazeMSS/OctoPrint-TopTemp #17: Add possibility to colorize the icons and labels

giacu92, 2021-01-26:
When using Dark settings on octoprint icons and label cannot be read as the gray is very similar to topbar

giacu92, 2021-01-26:
[image]

LazeMSS, 2021-01-26:
Well there is no color defined in my plugin. The theme should change default font colour. I will look later and maybe add an option to set label color and icon color

shadowpanther, 2021-01-29:
Proper colors for navbar elements seem to come from .navbar-text class. But just adding this class anywhere in your layout messes the navbar because the dimension it sets are suited for one-line elements.

So, I propose:
- add a class to the container:
```
<li id="navbar_plugin_toptemp" class="navbar-text" data-bind="allowBindings: true" style="margin-right: 4px; margin-left: 4px;">
```
- override the line-height back to the previous value (I don't know if if there is a better way to do this):
```
#navbar_plugin_toptemp > div {
	margin-top: -2px;
	line-height: 20px;
}
```

LazeMSS, 2021-02-02:
@giacu92 and @shadowpanther please try the latest release:

https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.0.3

I'm closing for now - if not fixed please reopen this :)

closed 2021-02-02 by LazeMSS

giacu92, 2021-02-02:
[image]

working flawlessly :) thanks!