LazeMSS/OctoPrint-TopTemp #18: Make all information from the M105 command available

fdraeger, 2021-01-31:
This request asks to provide a selection of temperatures returned from the M105 command.

I was trying to display the probe sensor temp which was not possible as custom temp via executing a shell command.

My printer (Caribou3D) provides:

`Send: M105
Recv: ok T:18.3 /0.0 B:18.6 /0.0 T0:18.3 /0.0 @:0 B@:0 P:19.3 A:29.2`

So this request is to provide a selection to allow to pick P (Probe) and A (Ambient) temperatures to be shown.

LazeMSS, 2021-01-31:
Im looking into it :)

fila612, 2021-02-02:
i'm also interested for this enhancement

LazeMSS, 2021-02-02:
I have it running in a test version now. Will release it after the next release.

LazeMSS, 2021-02-19:
This is fixed in the next release - please test/try by following:
```
Settings -> Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install
```

LazeMSS, 2021-02-21:
Closing - fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.0

Please reopen if not fixed

fila612, 2021-02-22:
thanks for that feature, on my system it works only for PINDA Probe, not for ambient.
settings for probe and ambient quite similar, but no values showing for ambient.
So I can't reopen this ticket, maybe someone else could do that, or I'm the only one with that error.

LazeMSS, 2021-02-22:
1. Have you tried turning of "Only show while printer is printing" if thats enabled?
2. With the data you have in the screenshot choosing "Type" = "From printer" and selecting the predefined types - they should both work.

fila612, 2021-02-22:
Hi, I've compared the PINDA and the ambient settings and they are mostly the same, except the regex to catch the values.
I've nothing changed and I'm using the predefinitions and I think also that they should work (the only difference is the "A" instead of "P"....
The option "Only show while printer is printing" was not set.

here are some samples of M105:
ok T:234.8 /235.0 B:75.0 /75.0 T0:234.8 /235.0 @:63 B@:29 P:62.5 A:50.3
ok T:234.8 /235.0 B:75.0 /75.0 T0:234.8 /235.0 @:62 B@:27 P:62.6 A:50.3
ok T:235.1 /235.0 B:75.0 /75.0 T0:235.1 /235.0 @:58 B@:29 P:62.5 A:50.2
ok T:234.8 /235.0 B:75.0 /75.0 T0:234.8 /235.0 @:65 B@:33 P:62.5 A:50.2
ok T:234.1 /235.0 B:75.0 /75.0 T0:234.1 /235.0 @:74 B@:30 P:62.3 A:50.3
ok T:234.0 /235.0 B:75.0 /75.0 T0:234.0 /235.0 @:73 B@:34 P:62.2 A:50.4
ok T:234.7 /235.0 B:74.9 /75.0 T0:234.7 /235.0 @:61 B@:43 P:62.1 A:50.3

LazeMSS, 2021-02-22:
Could you post a screenshot of the settings?

fila612, 2021-02-22:
sure
[image]
[image]

LazeMSS, 2021-02-22:
Try with this pattern:
`(?:T|T0):.*? A:([^ ]+)[^<>\r\n]`
I think I might have made a mistake - it includes the newline at the end :/

fila612, 2021-02-22:
thanks, that works!!! :) 👍 
[image]

LazeMSS, 2021-02-22:
Great - will fix in the next version - thanks for the feedback.

commit c88bfd74 (2021-02-22): Fixed #18
closed 2021-02-24 by LazeMSS