LazeMSS/OctoPrint-TopTemp #20: After the latest update, shows "Waiting..." in the top bar.

shadowpanther, 2021-02-02:
As it says in the title. I tried restarting Octoprint, refreshing the webpage, connecting the printer to get actual data, etc. Nothing updates.

[image]

LazeMSS, 2021-02-02:
Can you try and turn on your developer console (F12 in chrome windows ) and refresh and post the output from that?

shadowpanther, 2021-02-02:
```
Error calling onAllBound on view model TopTempViewModel : TopTempViewModel/self.onAllBound@http://raspistripes.local/static/webassets/packed_plugins.js?35f02421:31542:30
callViewModelIf@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:964:75
callViewModelsIf/<@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958:153
Pn@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:94753
ur/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:105184
callViewModelsIf@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958:95
callViewModels@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:957:72
bindViewModels@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1024:101
setTimeout handler*fetchSettings/<@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:726
c@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:28294
fireWith@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:29039
l@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:79800
o/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82254
EventHandlerNonNull*send@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82440
ajax@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:78223
OctoPrintClient.prototype.ajax@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:170
OctoPrintClient.prototype.get@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:424
OctoPrintSettingsClient.prototype.get@http://raspistripes.local/static/webassets/packed_client.js?8f609640:57:940
SettingsViewModel/self.requestData@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:245:48
fetchSettings@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:358
setTimeout handler*@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:1479
c@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:28294
fireWith@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:29039
l@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:79800
o/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82254
EventHandlerNonNull*send@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82440
ajax@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:78223
OctoPrintClient.prototype.ajax@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:170
OctoPrintClient.prototype.ajaxWithData@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:337
OctoPrintClient.prototype.post@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:625
OctoPrintClient.prototype.postJson@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:950
OctoPrintBrowserClient.prototype.passiveLogin@http://raspistripes.local/static/webassets/packed_client.js?8f609640:38:963
LoginStateViewModel/self.requestData@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:159:148
onServerConnect@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:1062
@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:1438
c@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:28294
fireWith@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:29039
Deferred/</s[t[0]]@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:30940
DataUpdater/self._onConnectMessage@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:838:434
OctoPrintSocketClient.prototype.propagateMessage/<@http://raspistripes.local/static/webassets/packed_client.js?8f609640:14:116
Pn@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:94753
ur/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:105184
OctoPrintSocketClient.prototype.propagateMessage@http://raspistripes.local/static/webassets/packed_client.js?8f609640:14:77
onMessage/<@http://raspistripes.local/static/webassets/packed_client.js?8f609640:21:316
Zt/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:103123
_t@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:97253
Vt/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:102923
ur/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:105191
onMessage@http://raspistripes.local/static/webassets/packed_client.js?8f609640:21:278
[5]</r.prototype.dispatchEvent@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:5788
[14]</</w.prototype._transportMessage/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:16405
[14]</</w.prototype._transportMessage@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:16356
[3]</i.prototype.emit@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:4669
packed_core.js:965:37
```

Full refresh output attached:
[console-export-2021-2-2_19-7-45.txt](https://github.com/LazeMSS/OctoPrint-TopTemp/files/5912588/console-export-2021-2-2_19-7-45.txt)

LazeMSS, 2021-02-02:
@shadowpanther are you on octoprint discord? Its hard to tell whats causing the problem - i would like a teamviewer or more info about the problem :)

shadowpanther, 2021-02-02:
I'm not on Octoprint discord, and I'd prefer not to give anyone direct access to my computer.

I can say that my Octoprint runs on RPi3, its web interface is behind haproxy (to pass the webcam through the same port) and I'm using Firefox, if that's relevant. Before updating to v0.0.0.3, TopTemp worked, besides the black-on-black issue.

LazeMSS, 2021-02-02:
I found the error - I forgot to test in Firefox :(

LazeMSS, 2021-02-02:
Fixed now :)

LazeMSS, 2021-02-02:
@shadowpanther please try and force an update using the advanced under "Octoprint -> Software update" in settings and report back.

shadowpanther, 2021-02-02:
Updated to 0.0.0.4, still shows "waiting"

[image]
The error message seems to be the same:

```
Error calling onAllBound on view model TopTempViewModel : TopTempViewModel/self.onAllBound@http://raspistripes.local/static/webassets/packed_plugins.js?264d5a07:31549:30
callViewModelIf@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:964:75
callViewModelsIf/<@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958:153
Pn@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:94753
ur/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:105184
callViewModelsIf@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958:95
callViewModels@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:957:72
bindViewModels@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1024:101
setTimeout handler*fetchSettings/<@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:726
c@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:28294
fireWith@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:29039
l@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:79800
o/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82254
EventHandlerNonNull*send@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82440
ajax@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:78223
OctoPrintClient.prototype.ajax@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:170
OctoPrintClient.prototype.get@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:424
OctoPrintSettingsClient.prototype.get@http://raspistripes.local/static/webassets/packed_client.js?8f609640:57:940
SettingsViewModel/self.requestData@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:245:48
fetchSettings@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:358
setTimeout handler*@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:1479
c@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:28294
fireWith@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:29039
l@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:79800
o/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82254
EventHandlerNonNull*send@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:82440
ajax@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:78223
OctoPrintClient.prototype.ajax@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:170
OctoPrintClient.prototype.ajaxWithData@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:337
OctoPrintClient.prototype.post@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:625
OctoPrintClient.prototype.postJson@http://raspistripes.local/static/webassets/packed_client.js?8f609640:7:950
OctoPrintBrowserClient.prototype.passiveLogin@http://raspistripes.local/static/webassets/packed_client.js?8f609640:38:963
LoginStateViewModel/self.requestData@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:159:148
onServerConnect@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:1062
@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1026:1438
c@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:28294
fireWith@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:29039
Deferred/</s[t[0]]@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:30940
DataUpdater/self._onConnectMessage@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:838:434
OctoPrintSocketClient.prototype.propagateMessage/<@http://raspistripes.local/static/webassets/packed_client.js?8f609640:14:116
Pn@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:94753
ur/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:105184
OctoPrintSocketClient.prototype.propagateMessage@http://raspistripes.local/static/webassets/packed_client.js?8f609640:14:77
onMessage/<@http://raspistripes.local/static/webassets/packed_client.js?8f609640:21:316
Zt/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:103123
_t@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:97253
Vt/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:102923
ur/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:105191
onMessage@http://raspistripes.local/static/webassets/packed_client.js?8f609640:21:278
[5]</r.prototype.dispatchEvent@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:5788
[14]</</w.prototype._transportMessage/<@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:16405
[14]</</w.prototype._transportMessage@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:16356
[3]</i.prototype.emit@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232:4669
packed_core.js:965:37
    callViewModelIf http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:965
    callViewModelsIf http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958
    Pn http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1
[... cut ...]
    ur http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1
    onMessage http://raspistripes.local/static/webassets/packed_client.js?8f609640:21
    dispatchEvent http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232
    _transportMessage http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232
    _transportMessage http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232
    emit http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1232
```

LazeMSS, 2021-02-02:
Try a force refresh (CTRL+F5) and did you restart octoprint?

shadowpanther, 2021-02-02:
Yes, Octoprint restarts itself for plugin updates, and I tried force-reloading(Cmd-Shift-R). I even tried to clear the cache.

LazeMSS, 2021-02-02:
Strange - I have no problems on Windows 10 with edge, chrome or firefox - have you tried deleting your toptemp settings (eraser icon in plugin manager)- there should be no problems migrating settings but maybe there is.

shadowpanther, 2021-02-02:
Just tried in Chrome - same result, even looks like the same error. Tried fully removing the plugin and settings and reinstalling it - same. Tried in Firefox and Chrome on Windows - same. Tried setting logging for toptemp to debug - nothing in the log besides enumerating ways to get system temperatures.

LazeMSS, 2021-02-02:
Wauww thats strange. But the error is 100% somewhere in the loading of the javascript.

What is on the line in the js:
Error calling onAllBound on view model TopTempViewModel : TopTempViewModel/self.onAllBound@http://raspistripes.local/static/webassets/packed_plugins.js?264d5a07:31549:30

and: 
callViewModelIf@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:964:75
callViewModelsIf/<@http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958:153
Pn@http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:94753

LazeMSS, 2021-02-02:
Oh and does the developer console say anything different in Chrome?

shadowpanther, 2021-02-02:
You mean, the line number from the error? packed_core.js:965:37
`if(raiseErrors){throw exc;}else{log.error("Error calling",method,"on view model",viewModel.constructor.name,":",exc.stack||exc);}}}`

Chrome's console says:

```
Error calling onAllBound on view model TopTempViewModel : TypeError: Cannot read property 'ctAxisTitle' of undefined
    at TopTempViewModel.self.onAllBound (http://raspistripes.local/static/webassets/packed_plugins.js?264d5a07:31549:54)
    at callViewModelIf (http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:964:75)
    at http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958:138
    at Pn (http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:94753)
    at Function.<anonymous> (http://raspistripes.local/static/webassets/packed_libs.js?ab739628:1:105184)
    at callViewModelsIf (http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:958:95)
    at callViewModels (http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:957:56)
    at bindViewModels (http://raspistripes.local/static/webassets/packed_core.js?b8ae0794:1024:87)
callViewModelIf @ packed_core.js?b8ae0794:965
```

LazeMSS, 2021-02-02:
Great - thanks :)

LazeMSS, 2021-02-02:
Fixed now I hope :)

LazeMSS, 2021-02-02:
@shadowpanther  try and update and confirm :)