LazeMSS/OctoPrint-TopTemp #22: CPU Usage Monitor

wlbillmartin, 2021-02-02:
Is it possible to add CPU Usage (Like the RasPi GUI CPU Usage Monitor) to TopTemp?

I'm REALLY happy with TopTemp, but I have to keep using VNC to look at the GUI to see my CPU Usage. I had a problem with OctoPrintWebcamStreemer that ran my CPU temp over 70 degrees and when I switched to VNC to look at the GUI my CPU was stuck at 100%. It would have been nice to see this on TopTemp.

BTW: I don't use the graphs, but the new PopUp feature is GREAT!!!

THANK YOU!!!

(Oh shit... I just noticed you did the UI Customizer... GREAT PLUGIN!!!)

LazeMSS, 2021-02-02:
Thanks :)

So what you want is the cpu usage in percentage - that is a planned features so yes.

wlbillmartin, 2021-02-02:
PERFECT!!!
 
Sir William Louis "Bill" Martin Jr., ATP, Capt. (Ret.), OMS
Active Participant - Planet Earth
AKA:
William P. Lewis
Comedian/Radio Personality/Actor
http://www.williamplewis.com/
Phone: 949.933.6230
Fax: 909.510.8253

LazeMSS, 2021-02-19:
This is fixed in the next release - please test/try by following:
```
Settings -> Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install
```

wlbillmartin, 2021-02-20:
My printer is down as I do some upgrades, but I LOVE UI Customizer & TopTemp!!! As soon as it’s up, I’ll check it out and get back to you.

Do you have a “Donate” area, I’d like to send some “Complement$”?
 
Sir William Louis "Bill" Martin Jr., ATP, Capt. (Ret.), OMS
Active Participant - Planet Earth
AKA:
William P. Lewis
Comedian/Radio Personality/Actor
http://www.williamplewis.com/
Phone: 949.933.6230
Fax: 909.510.8253

LazeMSS, 2021-02-20:
Thanks feel free to use PayPal: https://www.paypal.com/paypalme/ihccaptain

wlbillmartin, 2021-02-20:
DONE!
 
Sir William Louis "Bill" Martin Jr., ATP, Capt. (Ret.), OMS
Active Participant - Planet Earth
AKA:
William P. Lewis
Comedian/Radio Personality/Actor
http://www.williamplewis.com/
Phone: 949.933.6230
Fax: 909.510.8253

wlbillmartin, 2021-02-20:
Uninstalled TopTemp and installed development. Not getting Fan Speed (nice feature though!) (just a "searching") nor CPU Load. Bed Temp, Extruder Temp and CPU Temp working as per usual.

wlbillmartin, 2021-02-20:
Sent the command for Fan Speed (copied from Settings) from Terminal. This was output:
Send: ^M106.*?S([^ ]+)
Recv: echo:Unknown command: "^M106.

LazeMSS, 2021-02-20:
@wlbillmartin  no need to send anything - just select the cooling fan speed (its added as default) - but it won't show anything until you start a print - you can disable this by unchecking "Only show while printer is printing"

Then turn on the fan by pressing "Fan on" under control or send this command in the termnal:
`M106 S200`

This will turn on the fan of the printer for 200 - 255 is max. But normally is controlled while printing. But this way you can test.

If you want to see the cpu load then add a new custom and select type "Server data" and from the "Select" drodown select "CPU  usage percentage"

[image]

LazeMSS, 2021-02-21:
Closing - fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.0

Please reopen if not fixed

closed 2021-02-21 by LazeMSS

wlbillmartin, 2021-02-21:
I. AM. AN. IDIOT!

I never “played” with the custom controls, I thought I had to know code and stuff. Following your instructions… THERE IT IS!

THANK YOU!!!
 
Sir William Louis "Bill" Martin Jr., ATP, Capt. (Ret.), OMS
Active Participant - Planet Earth
AKA:
William P. Lewis
Comedian/Radio Personality/Actor
http://www.williamplewis.com/
Phone: 949.933.6230
Fax: 909.510.8253

 

image <https://user-images.githubusercontent.com/69049371/108603053-de38e200-73a5-11eb-9636-4141d908cd47.png>