LazeMSS/OctoPrint-TopTemp #24: No actual graph in the Graph Tooltip

lightmaster, 2021-02-02:
When I hover over any of the 3 charts that I have, the tooltip popup does show, but there's no graph within them.

[image]

LazeMSS, 2021-02-02:
Any errors in the console? Tried going to another item and back?

lightmaster, 2021-02-02:
I did try refreshing the page and doing a Ctrl+F5 refresh. How do I check the terminal?

LazeMSS, 2021-02-02:
F12 in chrome on windows

lightmaster, 2021-02-02:
Using Firefox

Not sure if its related, but I get `Waiting...` until I open up settings and the close settings. Both the TopTemp boxes and my webcam won't show up until I do that.

```
Uncaught TypeError: Chartist.plugins is undefined
    updatePopover https://example.com/octoprint/static/webassets/packed_plugins.js?2f57c6a4:64204
    buildContainers https://example.com/octoprint/static/webassets/packed_plugins.js?2f57c6a4:64037
    handle https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    dispatch https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    handle https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    add https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ke https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ke https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    on https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    buildContainers https://example.com/octoprint/static/webassets/packed_plugins.js?2f57c6a4:64026
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    buildContainers https://example.com/octoprint/static/webassets/packed_plugins.js?2f57c6a4:63991
    onSettingsHidden https://example.com/octoprint/static/webassets/packed_plugins.js?2f57c6a4:63423
    callViewModelIf https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:964
    callViewModelsIf https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:958
    Pn https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ur https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    callViewModelsIf https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:958
    callViewModels https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:957
    onAllBound https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:242
    dispatch https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    handle https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    trigger https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    trigger https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    trigger https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    hideModal https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:560
    hideWithTransition https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:559
    i https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    dispatch https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    handle https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    add https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ke https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ke https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    one https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    hideWithTransition https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:559
    hide https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:555
    modal https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:562
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    modal https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:562
    hide https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:242
    cancelData https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:256
    click https://example.com/octoprint/static/webassets/packed_libs.js?ab739628 line 3 > Function:3
    init https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:5
    dispatch https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    handle https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    add https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ke https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    each https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ke https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    on https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    B https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    init https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:5
[... cut ...]
    ur https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    bindViewModels https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:1016
    Pn https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    ur https://example.com/octoprint/static/webassets/packed_libs.js?ab739628:1
    bindViewModels https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:1007
    setTimeout handler*fetchSettings/< https://example.com/octoprint/static/webassets/packed_core.js?b8ae0794:1026
packed_plugins.js:64204:29
```

LazeMSS, 2021-02-03:
What version? I did a couple of panic releases 🙈😄

Latest is 0.0.0.5

lightmaster, 2021-02-03:
0.0.0.5, lol

LazeMSS, 2021-02-03:
Even more strange. I will looki again.

lightmaster, 2021-02-03:
May or may not be relavent, but I access my Octoprint through a https reverse proxy that is hosted on a server located at my house.

LazeMSS, 2021-02-03:
Should make no difference. It's loads a piece of JavaScript

LazeMSS, 2021-02-04:
I can't figure out why this is happening - it might be related to another plugin - does it work in other browsers than firefox?

LazeMSS, 2021-02-04:
Can you try and do an install of my development version -  go into Settings > Plugin Manager > Get More > ...from URL.
https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip

lightmaster, 2021-02-04:
I just started a 5 hour print, lol. I will after this gets done though.

lightmaster, 2021-02-04:
About to try installing the dev version. I did install chrome and try with chrome, and same symptoms: getting `Waiting...` until I press the settings button and then back out of settings (with or without saving anything), then no graph in the popup tooltip.

lightmaster, 2021-02-04:
The Dev version didn't help any

LazeMSS, 2021-02-04:
No I think why - im guessing with the reverse proxy.

Can you try and use chrome and look unde the network tab - look for it loading a file called "chartist" - click the "XHR" tab.

[image]

Can you post a screenshot of that?

lightmaster, 2021-02-04:
I disabled every 3rd party plugin, and ended up with `Waiting...` refusing to go away. Seems that something in TopTemp actually depends on UI Customizer to function, cuz `Waiting...` finally went away again after I enabled UI Customizer again.

LazeMSS, 2021-02-04:
Ahh so you have UI Customizer installed but disabled?

lightmaster, 2021-02-04:
UI Customizer installed but disabled: `Waiting...` that won't go away.

UI Customizer installed and enabled: `Waiting...` that goes away after opening and closing settings, and no graph in tooltip.

LazeMSS, 2021-02-04:
Strange

lightmaster, 2021-02-04:
[Pastebin for chartist saved as "HAR"](https://pastebin.com/cX7Q0tpn)

This seemed to be the most info I could get about that file from your screenshot.

LazeMSS, 2021-02-04:
@lightmaster please try installing the latest dev version again - hopefully its now fixed.

lightmaster, 2021-02-04:
At first, I didn't see any difference at all. Then I modified the octoprint startup script to allow direct access on all interfaces, and when going to http://ip:5000, I can now see the graphs and do not see `Waiting...` at all.

Not sure why it would make a difference, but on my installation, I have disabled HAProxy and use the built in nginx server as a reverse proxy intead of HAProxy, cuz it's so much faster to load this way. So various methods off access I've tried are:

External Server's Nginx Reverse proxy => OctoPi's Nginix Reverse Proxy => http://localhost:5000 and this doesn't work for graphs and shows `Waiting...`

OctoPi's Nginx Reverse Proxy => http://localhost:5000 Same results as first hitting the External Server's Reverse Proxy

Direct access to OctoPi via http://ip:5000 and This works like a charm.

LazeMSS, 2021-02-04:
Is this on the latest develop - I'm wondering if its caused by something weird. I will try on my installation but the reverse proxy should be able to run xhr data.

lightmaster, 2021-02-04:
i redownloaded the dev version when you said to a couple messages ago

lightmaster, 2021-02-04:
Seems to load other XHR data, if I'm understanding this network tab right. Also, I don't see `chartist` on there at all now

[image]

LazeMSS, 2021-02-04:
Try using the js tab instead

lightmaster, 2021-02-05:
think I found something. When accessing from a reverse proxy, it's calling `https://example.com/plugin/toptemp/static/js/chartist-plugin-axistitle.min.js` instead of `https://example.com/octoprint/plugin/....`

lightmaster, 2021-02-05:
If I got to `https://example.com/octoprint/plugin/toptemp/static/js/chartist-plugin-axistitle.min.js`, I get the JS file like I'd expect to

lightmaster, 2021-02-05:
Upon further investigation, I believe this is the same issue as Kragrathea/OctoPrint-PrettyGCode#29. If I go into `/home/pi/oprint/lib/python3.7/site-packages/octoprint_toptemp/static/js/TopTemp.js` and change line 840 from `/plugin/toptemp/static/js/chartist-plugin-axistitle.min.js` to `./plugin/toptemp/static/js/chartist-plugin-axistitle.min.js`, then it immediately loads when the page OctoPrint loads without getting stuck on `Waiting...` and the chart shows up.

lightmaster, 2021-02-05:
I replaced all 3 instances of `script.src = '/plugin/toptemp/...` within that file with `script.src = './plugin/toptemp/...` and it appears to work fine too. Looks the same as when I had replaced just the bit from the last message, so idk if it even changed anything at all with the other 2 sets of `script.src = '/plugin/toptemp/...`

LazeMSS, 2021-02-05:
Thanks I will look.

The example.com sounds like something wrong with you nginx configuration

LazeMSS, 2021-02-19:
This is fixed in the next release - please test/try by following:
```
Settings -> Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install
```

lightmaster, 2021-02-20:
I did an `uninstall and clean data` to ensure than it was installed fresh (within my temp modification to fix this surviving) and then installed the dev version. I can see the graphs on both direct access via ip:port and via reverse proxy.

Side note, is there supposed to be labels on the `X` axis? I see labels on the `Y` axis, but just a `Minutes` title with no numbers on the `X` axis.

LazeMSS, 2021-02-20:
There should numbers on the X axis. Post a screenshot 🤗

lightmaster, 2021-02-20:
[image]

LazeMSS, 2021-02-20:
Once again - how do you do this? :)

lightmaster, 2021-02-20:
> Once again - how do you do this? 😄

Me, on the right 🤣
[The art of testing.mp4](https://cdn.discordapp.com/attachments/217253743668756480/803934623355895828/The-art-of-testing.mp4)

And I'll see if console is complaining about anything else.

lightmaster, 2021-02-20:
I don't see anything that jumps out as pertaining to the graphs. Tried to look at them in the inspector window to see if the text was showing but invisible/color of background or too small, but the tooltip disappears as soon as I move my mouse off it, and that block of code is constantly changing with the graph updates, so as soon as I use the keyboard to scroll down to that code, it is rewritten which kicks the selected line back to the top of the html code for some annoying reason.

LazeMSS, 2021-02-20:
Try running this from the console
`OctoPrint.coreui.viewmodels.topTempViewModel.updatePaused = true;`

And rightclick on the popover to keep it there and move the mouse to console. That how I do it when its acting up.

lightmaster, 2021-02-20:
I see the numbers in the inspect window, not sure why they aren't showing up

lightmaster, 2021-02-20:
I disabled Themify and tried a couple different themes of UI Customizer, still aren't showing up

lightmaster, 2021-02-20:
Installed Chrome and it works in Chrome, so must be an issue with using Firefox.

LazeMSS, 2021-02-21:
Fixed :)

lightmaster, 2021-02-21:
Install the dev archive again to test?

LazeMSS, 2021-02-21:
Closing - fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.0

Please reopen if not fixed

closed 2021-02-21 by LazeMSS

lightmaster, 2021-02-21:
I see them now. Thanks.