LazeMSS/OctoPrint-TopTemp #25: Waiting stuck still persist

luco85, 2021-02-04:
Hi! After last update the error still persist. It work from first connection on my printer. After page refrash (and cache) it stuck in waiting message

LazeMSS, 2021-02-04:
Okay - can you try and post what your developer console says (F12 in chrome) and post that?

luco85, 2021-02-04:
I use safari. Do you know the console equivalent command? I’m not at home and now printer is working but i can try to give you this information by vpn. When printer will finish i can install that version

LazeMSS, 2021-02-04:
https://support.apple.com/guide/safari/use-the-developer-tools-in-the-develop-menu-sfri20948/mac

luco85, 2021-02-04:
[Log] Create PrettyGCode View Model (packed_plugins.js, line 17128)
[Info] Starting dependency resolution... (packed_core.js, line 850)
[Info] ... dependency resolution done (packed_core.js, line 863)
[Info] Initial application setup done, connecting to server... (packed_core.js, line 886)
[Info] Connected to the server (packed_core.js, line 698)
[Info] Triggering connect on all view models (packed_core.js, line 703)
[Info] Server (re)connect processed (packed_core.js, line 705)
[Info] Finalizing application startup (packed_core.js, line 886)
[Info] Going to bind 55 view models... (packed_core.js, line 867)
[Info] Did not bind view model – "TimelapseViewModel" – "to target" – "#timelapse" – "since it does not exist" (packed_core.js, line 879)
[Info] Did not bind view model – "TimelapseViewModel" – "to target" – "#timelapse_link" – "since it does not exist" (packed_core.js, line 879)
[Info] Did not bind view model – "BackupViewModel" – "to target" – "#wizard_plugin_backup" – "since it does not exist" (packed_core.js, line 879)
[Info] Did not bind view model – "UsageViewModel" – "to target" – "#wizard_plugin_tracking" – "since it does not exist" (packed_core.js, line 879)
[Info] Did not bind view model – "ConnectionViewModel" – "to target" – "#connection_wrapper" – "since it does not exist" (packed_core.js, line 879)
[Info] Did not bind view model – "SoftwareUpdateViewModel" – "to target" – "#softwareupdate_confirmation_dialog" – "since it does not exist" (packed_core.js, line 879)
[Info] Did not bind view model – "SoftwareUpdateViewModel" – "to target" – "#wizard_plugin_softwareupdate" – "since it does not exist" (packed_core.js, line 879)
[Info] Did not bind view model – "bedlevelvisualizerViewModel" – "to target" – "#wizard_plugin_bedlevelvisualizer" – "since it does not exist" (packed_core.js, line 879)
[Log] onAfterBinding (packed_plugins.js, line 17649)
[Info] User Luco logged in (packed_core.js, line 159)
[Log] Should show widescreen?: true (packed_plugins.js, line 27462)
[Info] ... binding done (packed_core.js, line 884)
[Info] Application startup complete (packed_core.js, line 884)
[Error] Failed to load resource: the server responded with a status of 404 (NOT FOUND) (undefined, line 0)
[Warning] The resource http://192.168.1.254/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.1.254/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.1.254/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.1.254/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.1.254/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.1.254/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Error] Failed to load resource: the server responded with a status of 404 (NOT FOUND) (chartist.min.js.map, line 0)
[Error] Failed to load resource: the server responded with a status of 404 (NOT FOUND) (chartist-plugin-axistitle.min.js.map, line 0)

LazeMSS, 2021-02-04:
Nothing in there indicates an error - i will try on my mac later today

luco85, 2021-02-04:
Ok thanks! With version 0.0.0.2 it worked fine for me. From 3 i had problem

Inviato da iPhone

luco85, 2021-02-04:
One question
Is possible to change font’s color on top bar? Thanksfor your time

Inviato da iPhone

Dtuxcomp, 2021-02-04:
Hi, I'm new to GitHub so if this is in the wrong place or whatever let me know.

I've also got this issue, it started with the update a few days ago, that caused it to get stuck on waiting . . .  but only on the OctoPi instance on my Ender 3 v1, my other printer is an Ender 3 Pro and that carried on working fine, then today's update caused the same issue there. I have tried removing the app data with the cleanup option, rebooting the Pi, removing and reinstalling the app. Both Pi's are 3B's running fully updated versions of OctoPi.

Thanks,

Dave.

LazeMSS, 2021-02-04:
Really strang - @Dtuxcomp  what browser and computer etc.?

Can you try and do an install my development version go into Settings > Plugin Manager > Get More > ...from URL.
https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip

Dtuxcomp, 2021-02-04:
Hi, I'm using the Firefox browser v.85.0 on Linux Mint XFCE 20.1 on a Dell Precision T5600. OctoPi is updated and the OctoPrint version is 1.5.3 the current Top Temp is v 0.0.0.5 - I first had this issue on the Ender 3 v1 at v 0.0.0.4 and the update to v 0.0.0.5 today brought the same issue to the Ender 3 Pro which is running the latest firmware from TH3D, the other Ender v1 is still on the stock firmware.

Thank you for your time and help with this, I'll now try the dev version as you suggest and let you know what happens.

Dave.

Dtuxcomp, 2021-02-04:
Just tried the dev version and that's not changed anything sadly, it still sticks on saying "waiting . . ."

LazeMSS, 2021-02-04:
@Dtuxcomp  do you have any other plugins installed? Running reverse proxy etc?

Dtuxcomp, 2021-02-04:
No I don't think so, running fairly stock, and there's no proxy as it's all on my LAN. Can I just download v0.0.0.3 / 0.0.0.4 directly for now as that was working ?

LazeMSS, 2021-02-04:
Yes you can :) its on the github page. I will close this then and make an update asap.

closed 2021-02-04 by LazeMSS