LazeMSS/OctoPrint-TopTemp #26: Feature Request: Sensing while offline

SuperNose, 2021-02-13:
I've put a sensor (AM2302) inside my printer, to monitor the temperature. Works super, but your plugin won't show me the temperature until I turn on the printer - otherwise, it just shows "waiting". After switching the printer ON and later OFF, your plugin keeps showing then correctly those CPU and mine "Inside printer" measurement.

New feature request: the sensors should have a "Wait for printer connection" bool/checkbox and those isPrinterOnlineNeeded==false should be always visible in the bar, showing the always available sensor values.

Btw: it would be probably quite practical for the others to have some default support for the AM2302, to watch different things, like the printer-inside temp, or PSU-inside temp, or room-temp etc.. It's just a simple sensor with a 10k resistor between the pins 1 (5V+) and 2 (signal). 

Thank you very much!

LazeMSS, 2021-02-20:
This is fixed in the next release - please test/try by following the below guide:

Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install

LazeMSS, 2021-02-20:
And post separate request for new features 🤗

LazeMSS, 2021-02-21:
Closing - fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.0

Please reopen if not fixed

closed 2021-02-21 by LazeMSS

SuperNose, 2021-02-22:
Wow, that was fast, thank you 😁!