LazeMSS/OctoPrint-TopTemp #27: Feature Request: Allow Themeing

MasterDomino, 2021-02-15:
Hello,

I would like to request a feature added, as i tried to manually add changes to themeify but that wont work as the script executes after themeify enforces its rules .popover-title and .popover-content cannot be edited.

while writing this i have in mind that the css follow other theme settings set by themeify by reading what theme is enabled.

Im thankful for consideration.

LazeMSS, 2021-02-15:
I will try - have you tried adding !important to your custom css? :)

MasterDomino, 2021-02-15:
Have just tried that and it doesnt seem to work, i dont really know if it's themeifies fault or me being a dummy, also thanks for the great plugin really adds readability

LazeMSS, 2021-02-15:
Im working on implemeting my own themes :)

MasterDomino, 2021-02-24:
this works beautifully with UI Customizer plugin making Themeify kinda obsolete, i guess this can be closed(?)

closed 2021-02-24 by LazeMSS