LazeMSS/OctoPrint-TopTemp #28: Feature request: Read Pi temperature sensors

bladeslapper, 2021-02-20:
I have the latest development version (it looks great) and thought it might be a really nice addition to read temp sensors on the PI GPIO pins. I currently use the Enclosure Plugin with some DB18B20 sensors but TopTemp could possibly do away with the need for this plug-in?

LazeMSS, 2021-02-21:
If you want to make support for hardware then I need that hardware - its impossible to code in the blind. 

Feel free to donate: https://www.paypal.com/paypalme/ihccaptain so I can buy the hardware and support it.

I don't like relying on other plugins - if they change my stuff breaks etc.

bladeslapper, 2021-02-21:
That's fair enough.
I've just sent you some money. I use the DS18B20 sensor although the enclosure plug-in supports 10 different types all relatively low cost sensor and I'm not sure which are the most popular.
If you get the chance to use it to buy some sensors and develop the plug-in then great but if not then just treat it as a donation towards the project.
keep up the good work.
B.

LazeMSS, 2021-02-21:
Thanks :) I will look.

bladeslapper, 2021-02-23:
> I don't like relying on other plugins - if they change my stuff breaks etc.

By the way, I agree with this however you may find it useful to know that the Dashboard plugin (https://github.com/j7126/OctoPrint-Dashboard#dependencies) integrates with the Enclosure plugin to draw in data from these temperature sensors

LazeMSS, 2021-02-23:
I know but then again this would require users to install a plugin to use my plugin which I really don't like that.

And users need to configure stuff in another plugin but on the downside then my plugin would block for that.

I think I will make both options: detect the enclosure plugin and show them as options and also add a direct option to add them inside my plugin.

Hardware has been ordered 💪

bladeslapper, 2021-02-23:
Yes I know exactly what you mean and I'm not sure what the answer is - you have the challenge of that decision ;-)

The enclosure plugin is very powerful and I would suspect that most people who have additional sensors are taking advantage of its features. I even use it to turn my printer on! The dashboard plugin seems to draw in data from a few different plugins. It works fine without this extra data but bringing it all together in a single place is a nice convenience.

Good luck!

flavict, 2021-02-23:
I do use 1 DS18B20 to monitor the room temperature. What I did was to create a custom setting and using this command:

find /sys/bus/w1/devices/ -name "28-*" -exec cat {}/w1_slave \; | grep "t=" | awk -F "t=" '{print $2/1000}'

to pull the value. If you have multiple sensors you may have to be more specific with the "28-*" in the command.

LazeMSS, 2021-02-23:
@flavict  awesome - yearh as far as I can tell you could basically do a cat of the exact device id:
cat  /sys/bus/w1/devices/28-00000674869d/w1_slave | grep "t=" | awk -F "t=" '{print $2/1000}'

That was actually how I was looking into doing it based on: https://www.waveshare.com/wiki/Raspberry_Pi_Tutorial_Series:_1-Wire_DS18B20_Sensor

So basically there is no need to make support for it - its already there by the power bash :) I could of course detect any devices matching this pattern on startup like the other stuff i detect.

LazeMSS, 2021-02-24:
@bladeslapper  can you test the way @flavict use it?

bladeslapper, 2021-02-24:
> 
> 
> @bladeslapper can you test the way @flavict use it?

Sorry I'm not sure of the best way to do that but if you can give me some more instructions I'd be happy to.
At the moment, I have 3 sensors all connected on the same GPIO and all fed in to the Enclosure plugin. One of them is used to trigger an emergency printer shutdown. They identify in the plugin using the 28-xxx reference.

flavict, 2021-02-24:
This is how it is setup in my OctoPrint.

[image]

flavict, 2021-02-24:
test with full "name" of the device...

[image]

LazeMSS, 2021-02-24:
@flavict  does this not work:
`cat /sys/bus/w1/devices/28-0316a184eff/w1_slave | grep "t=" | awk -F "t=" '{print $2/1000}'`

The above should be less hard on the system ressource when you know the exact name of the device

flavict, 2021-02-24:
no

[image]

LazeMSS, 2021-02-24:
I mispelled your device id 👅 
`cat /sys/bus/w1/devices/28-0316a184e0ff/w1_slave | grep "t=" | awk -F "t=" '{print $2/1000}'`

cat /sys/bus/w1/devices/**[INSERTIDHERE]**/w1_slave | grep "t=" | awk -F "t=" '{print $2/1000}'

flavict, 2021-02-24:
yes it work with: 

cat /sys/bus/w1/devices/28-0316a184e0ff/w1_slave | grep "t=" | awk -F "t=" '{print $2/1000}'

[image]

bladeslapper, 2021-02-24:
> Are you using the Enclosure plugin to trigger Raspberry IO output? Are you using the "Label" as the trigger.

Yes exactly that. Set 1 output to trigger IO used for a relay / label it Power Switch / auto-turn on during boot / hide UI.
Then set another output as a temperature alarm which references one of the "input" DS18B20 and "Control IO" = Power Switch.
If the temperature threshold is broken it turns the relay off and kills the printer. I have this sensor mounted in the main wire bundle where it goes into the print head.

I was originally intending to have "Temperature Alarms" set for all 3 of my sensors but for some reason when I set it up like that, the additional ones no longer reported temperatures in OctoPrint (they still work in PuTTY.) I've not really explored why.  The other two sensors are on the printer main board and inside the enclosure. I was hoping to use these 3 to detect a serious problem and kill the power. The one in the head is the most important though.

I thought that TopTemp would give me a really nice way of easily monitoring these temperatures from every page

bladeslapper, 2021-02-24:
> 
> 
> cat /sys/bus/w1/devices/[INSERTIDHERE]/w1_slave | grep "t=" | awk -F "t=" '{print $2/1000}'

@LazeMSS - Yep that works perfectly for me as well

[image]
.

LazeMSS, 2021-02-28:
I haven't give up on this 👍 There will be native/better support thanks to @bladeslapper  donation 🤗

Zilveron, 2021-03-05:
Similar issue, I think.  Mainly lack of knowledge on my part.  Been struggling with reading a DHT11 and was wondering what command I would send to read temperature or humidity on a GPIO pin.  I'll be getting sensors later on that work better with a Raspberry Pi but just want to work on a proof of concept at the moment.  The intent being to put a sensor in each of my dry boxes to display the humidity and to warn me when the humidity rises above a certain point and it is time to recharge the desiccant.  The Adafruit libraries are installed and I can run a Python script that reads the sensor, though not reliably.  I understand that the DHT sensors don't work great with the Pi but it is what I have on hand.  Thank you.

LazeMSS, 2021-03-05:
https://peppe8o.com/using-raspberry-pi-with-dht11-temperature-and-humidity-sensor-and-python/

But just change the code to be:
`import time
import board
import adafruit_dht
#Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT11(board.D17)
  try:
       # Print the values to the serial port
       temperature_c = dhtDevice.temperature
       temperature_f = temperature_c * (9 / 5) + 32
       humidity = dhtDevice.humidity
       print("Temp: {:.1f} F / {:.1f} C    Humidity: {}% "
             .format(temperature_f, temperature_c, humidity))
  except RuntimeError as error:     # Errors happen fairly often, DHT's are hard to read, just keep going
       print(error.args[0])
`

Then you can run this as a custom command.

Zilveron, 2021-03-05:
Took a bit of tweaking (and learning) on my part but I got it working and doing what I wanted it to do.  Programming languages are not my thing.  I never got past GW Basic way back in the mid 80's.

Thank you very much for the help.  I appreciate it and the work you have done in making this plugin.  Its a good one.

LazeMSS, 2021-04-06:
@bladeslapper @flavict the latest release(https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.4) has native support for the DS18B20 sensor.

Please test :)

Also added a short guide: https://github.com/LazeMSS/OctoPrint-TopTemp/wiki/How-to-setup-1-Wire-DS18B20-Sensor

flavict, 2021-04-06:
Tested and working. Thank you

[image]

bladeslapper, 2021-04-08:
Sorry for the delay.
Excellent yes that works for me as well.
I see that you have added it both under "Server Data" and also "Command"
Presumably it makes no difference?

bladeslapper, 2021-04-08:
> Also added a short guide: https://github.com/LazeMSS/OctoPrint-TopTemp/wiki/How-to-setup-1-Wire-DS18B20-Sensor

This looks excellent too

LazeMSS, 2021-04-09:
> Sorry for the delay.
> Excellent yes that works for me as well.
> I see that you have added it both under "Server Data" and also "Command"
> Presumably it makes no difference?

Ahhh you mean the "Temperature w1_slave_temp-1 Test Select " - doohh apparently psutil automatically detects and add the 1 wire temp sensor - I didn't even check that. So basically I didn't have to do antyhing 👅

bladeslapper, 2021-04-10:
Bugger!!
Yes I mean that. Although are you certain that was there before? I had a really good look around the plugin 2 months ago so it would be annoying if it turns out I missed that!

Either way, it works well:
[image]

Bendermk6, 2021-04-11:
Hi,
so i got two DS18B20 sensors as water-proof version like Robert Elder (https://blog.robertelder.org/ds18b20-raspberry-pi-setup-pullup/), only 3-wires and no pull-up resistor on it. I tried half of my Saturday to get the sensor working with not that much success.

I tried the method from Robert without the resistor (because i don't have any^^) -> +3V3, Ground and GPIO 16.

Kernel modules are active and still mounted after rebooting, thanks to Roberts tutorial:
w1_gpio                16384  0
w1_therm               24576  0
wire                   36864  2 w1_gpio,w1_therm

The pull-up in config with
`dtoverlay=w1-gpio-pullup,gpiopin=16` 
did not work, after reboot the GPIO16 is still 0V.
**Any idea why?**

Then i tried it with the small script to pull-up GPIO16 like Robert:
`#!/usr/bin/python
import RPi.GPIO as GPIO

GPIO_PIN_NUMBER=16
GPIO.setmode(GPIO.BCM)
GPIO.setup(GPIO_PIN_NUMBER, GPIO.IN, pull_up_down=GPIO.PUD_UP)`
that worked and changed GPIO16 to ~3V and i got finally some measurements from the sensor with:
`cat /sys/bus/w1/devices/28-xxxxxxx/w1_slave`

But after few minutes it stops to measure, like there is no output at all when i try
`cat /sys/bus/w1/devices/28-xxxxxxx/w1_slave`

Is it possible that i need a pull-up resistor? -_-

I mean i can donate you my second sensor, if you like somewhere in the EU here^^

Thanks.

flavict, 2021-04-11:
@Bendermk6 
I am using the same water-proof sensor.

You will need that resistor. You can use a value of 4,7kohms between 3,3V and data

There is two ways to connect that sensor.

Normal power mode:
[image]

and parasite power mode:
[image]

I am using the parasite power mode since it was the only way to make it work.

and my config.txt is done with this line: dtoverlay=w1-gpio-pullup,gpiopin=4,pullup=1,extpullup=5

The default GPIO pin for w1 is 4 (https://www.waveshare.com/wiki/Raspberry_Pi_Tutorial_Series:_1-Wire_DS18B20_Sensor)

Hope this help

Bendermk6, 2021-04-11:
@flavict
Thank you. I had to search my fathers cabinets full of old resistors und found after a while 2,4kohm ones.
Soldered two together and got something ~4.8kohm.

Now it works on the first try!

LazeMSS, 2021-04-12:
@flavict maybe we should include this in the wiki:
https://github.com/LazeMSS/OctoPrint-TopTemp/wiki

flavict, 2021-04-15:
@LazeMSS yup

closed 2022-12-03 by LazeMSS