LazeMSS/OctoPrint-TopTemp #31: Error daemon=True

GitterKEIS, 2021-02-24:
Great app, nice and functional without any other comments
Function to add custom sensor works fine....
Running version : Top Temp (0.0.1.0)

Have check the octo log...and after that checked your ini file
do not see any imports leading to line or argument...
Can you clear this out how it was original meant...

I have posted here belog the error....
Anyhow, great work, if I have some time left I will look further in it...Cheers!

2021-02-24 01:26:37,273 - octoprint.plugin - ERROR - Error while calling plugin toptemp
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_toptemp/__init__.py", line 113, in on_after_startup
    threading.Thread(target=self.gcodeRecvQworker, daemon=True).start()
TypeError: __init__() got an unexpected keyword argument 'daemon'

LazeMSS, 2021-02-24:
Try the dev version:
`Open Settings, then click Plugin Manager
At the bottom click "Get more.."
In the input field below "... from URL" paste in: https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip
Click Install`

LazeMSS, 2021-02-24:
Closing - is now released.

closed 2021-02-24 by LazeMSS