LazeMSS/OctoPrint-TopTemp #33: Feature request: Check Pi power supply state

Alex-Goaga, 2021-02-25:
Hello,
This plugin is absolute fantastic and the ability to customize your own commands and i was wondering if you can implement a default command like the get CPU template running at 30 seconds for the following :

/opt/vc/bin/vcgencmd get_throttled|cut -d "=" -f2|cut -d"'" -f1   <- this command returns bits and hex values.
For example : If the answer is 0x0 , Pi is good with the supplied voltage and SoC temperature.

function throttleCodeMask {
  perl -e "printf \"%s\", $1 & $2 ? \"$3\" : \"$4\""
}

function throttledToText {
  throttledCode=$1
  throttleCodeMask $throttledCode 0x80000 "Soft temperature limit has occurred, " ""
  throttleCodeMask $throttledCode 0x40000 "Throttling has occurred, " ""
  throttleCodeMask $throttledCode 0x20000 "Arm frequency capping has occurred, " ""
  throttleCodeMask $throttledCode 0x10000 "Under-voltage has occurred, " ""
  throttleCodeMask $throttledCode 0x8 "Soft temperature limit active, " ""
  throttleCodeMask $throttledCode 0x4 "Currently throttled, " ""
  throttleCodeMask $throttledCode 0x2 "Arm frequency capped, " ""
  throttleCodeMask $throttledCode 0x1 "Under-voltage detected, " ""
  throttleCodeMask $throttledCode 0x0 "Perfect Supplied, " ""
}

 throttled_text=$(throttledToText $(vcgencmd get_throttled | cut -f2 -d=))

OR , a button to accept the "error state" given by the command as its not a "value"

LazeMSS, 2021-02-25:
But octopi already had this built-in in so why make it again?

LazeMSS, 2021-02-25:
If you don't think that is a good solution then reopen and lets discuss why the native solution is not good.

closed 2021-02-25 by LazeMSS

Alex-Goaga, 2021-02-25:
Sorry for the late reply.
I've seen the octoprint native solution for this probleme, but i never encounter it and i dont know how it manages to display / alert the end user and when.

My idea was for a "heart-beat" check every minute as in case the reading / or the issue can be found in an early stage before the octo print server may encounter it.

LazeMSS, 2021-02-25:
This is implemented - if you want to "fake it" then run this from a developer console:
`$('#navbar_plugin_pi_support *').show();`

Also check Settings -> Octoprint -> Pi Support