LazeMSS/OctoPrint-TopTemp #35: Feature request: Add support for Klipper temperature sensors

Fank, 2021-03-03:
Some other frontends support klipper temperature sensors, which you can define in the Klipper config:
```
[temperature_sensor mcu_temp]
sensor_type: temperature_mcu
min_temp: 0
max_temp: 100

[temperature_sensor raspberry_pi]
sensor_type: temperature_host
min_temp: 10
max_temp: 100
```

- https://docs.mainsail.xyz/quicktips/additional-sensors
- https://docs.fluidd.xyz/features/sensors

So far i have no clue how this exactly works, but it looks like you can access the temperature with `printer["temperature_sensor mcu_temp"].temperature` they were talking about it [here](https://github.com/KevinOConnor/klipper/issues/3798#issuecomment-765235324)

LazeMSS, 2021-03-04:
You can make your own patterns for this using the custom filters/regexp - no need to add direct support for this as far as I can see.

Maybe look here: https://github.com/LazeMSS/OctoPrint-TopTemp/issues/16

Fank, 2021-03-05:
The plugin would need to query this, since there are hidden in some variables. There are no gcodes or something for easily access them.

Here is a example of how [Fluidd](https://github.com/cadriel/fluidd) access them:
https://github.com/cadriel/fluidd/blob/7374fdbed9dfe0e5824d5ea5869a2aacfc5c0a59/src/store/socket/getters.ts#L485-L493

Like shown here https://github.com/KevinOConnor/klipper/blob/master/docs/Command_Templates.md it would be nice if the plugin could check is a key with prefix `temperature_sensor ` is available show them in a dropdown and make them able to use.

So far it looks like you also write a custom gcode macro to be able to access this variable, this is also a solution but something native from this plugin would be much better.

LazeMSS, 2021-03-05:
Yes but you need to send a command to the printer to get these values and I really dont want to do this because this might disturb printing etc.

LazeMSS, 2021-04-06:
Closing - i can't get klipper running for now - if someone want this they should make a PR

closed 2021-04-06 by LazeMSS