LazeMSS/OctoPrint-TopTemp #36: Feature request: Add min/max/avg to charts

Fank, 2021-03-03:
In example for the CPU we have "Current"
[image]

It would be nice to see the min/max/avg (chooseable) as well, for the time duration seeing in the chart.

LazeMSS, 2021-03-05:
The time duration is kinda hard - if not fixed and im trying not to flood the mem and apis with data etc.

Im also trying to declutter the settings so instead I will make it so it set the low near the lowest read instead of 0

LazeMSS, 2021-03-05:
Once I rewrite the settings into a more modular form I will revisit this and probally implement full config of the graphs.

LazeMSS, 2022-12-11:
Should be fixed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.0 - please reopen if not working/fixed as expected

closed 2022-12-11 by LazeMSS

puterboy, 2022-12-24:
Not sure if this is a bug or intended behavior but the min/max seem to be calculated over the entire "uptime" of Octoprint while the associated graph is limited to the specified timescale. This is a bit confusing, especially if you have long uptimes...

LazeMSS, 2022-12-25:
> Not sure if this is a bug or intended behavior but the min/max seem to be calculated over the entire "uptime" of Octoprint while the associated graph is limited to the specified timescale. This is a bit confusing, especially if you have long uptimes...

Yes the graph is sliced for the period and the min/max in done on the entire dataset. Basically its meant to be like that because the small overlay graph is only meant to show what you have specified now

puterboy, 2022-12-26:
I get that.
I wasn't referring to the icon overlay graph but to the hover/pop-up graph. Since you can set the time duration for the pop-up graph, it would seem logical that the min/max/avg should apply to that time period which can be as short or as long as you like -- this is almost implied since the min/max/avg appear on that same pop-up graph and one would expect some correlation.

Indeed, the way I discovered this issue is when I noted that the min/max/avg noted on the graph had nothing to do with the actual graph showing which was quite weird.

Conversely, I keep Octopi running continuously and only reboot it when I need to upgrade modules. Any min/max/average over that period is almost meaningless. Especially since the period of that min/max/average is not displayed and unknowable unless you do something like a 'ps' on Octoprint to see when it was launched...

Just my 2 cents :)

LazeMSS, 2022-12-27:
> I get that. I wasn't referring to the icon overlay graph but to the hover/pop-up graph. Since you can set the time duration for the pop-up graph, it would seem logical that the min/max/avg should [...]

Fixed in next