LazeMSS/OctoPrint-TopTemp #37: Feature request: Themify charts

Fank, 2021-03-03:
It would be nice to use the default theme or custom define the colors for the charts.
In my example i am using the plugin [UI Customizer](https://github.com/LazeMSS/OctoPrint-UICustomizer/) and it looks like:
[image]

Would be nice if it could use the same colors as in the temperature tab, if default theme is choosen:
[image]

LazeMSS, 2021-03-03:
Well this requires the theme authors to implement this feature - i will look into it.
I also make the UI Customizer plugin 😄

Fank, 2021-03-03:
I know :P

closed 2021-03-05 by LazeMSS

LazeMSS, 2021-03-05:
Now released

Fank, 2021-03-05:
You should update the pic in the Readme too :D

LazeMSS, 2021-03-05:
Done...