LazeMSS/OctoPrint-TopTemp #39: Change text color?

Skiddz, 2021-03-03:
Potentially stupid question, or feature request.  :)

Is it possible to change the text color of the display?  I run two printers from one Pi and each printer's Octoprint instance is a different color.  One of my printers uses a blue info bar and the grey text is pretty much impossible to see.  I'd like to make it white if possible.  Is there something I can tweak or can a text color selection be added to the Custom tab?

LazeMSS, 2021-03-04:
I will look into it.

Skiddz, 2021-03-04:
Much appreciated!

Skiddz, 2021-03-05:
Potentially stupid question, the sequel.
Would a quick edit to the style sheet do the trick?

I assume the correct file is:
~/oprint/lib/python3.7/site-packages/octoprint_toptemp/static/css/TopTemp.css

and I'm guessing this section is where color of the text might be settable?

#navbar_plugin_toptemp div.TopTempText {
    height: 100%;
    margin-top: 6px;
    position: relative;
    line-height: initial;
    text-align: center;

LazeMSS, 2021-03-05:
Yes - easier to add a bit of custom css on
#navbar_plugin_toptemp.navbar-text

But I will make support for the different native colors

LazeMSS, 2021-03-05:
Try the develop version:
https://github.com/LazeMSS/OctoPrint-TopTemp/archive/develop.zip

Skiddz, 2021-03-05:
I'll give that a try as soon as my current print job finishes.  Thx!

closed 2021-03-05 by LazeMSS

LazeMSS, 2021-03-05:
@Skiddz  no need I just released it into the wild - but please comment is still broken :)

Skiddz, 2021-03-05:
Hey!  I can see the text now!  Thanks!

Skiddz, 2021-03-08:
So this is interesting..  I run 2 instances of OctoPrint on one Pi.  In the "main" instance, the color tweak works.  I can edit the .css to change color as I see fit.  In the 2nd instance, the text is still grey.  I haven't had the time to really dig in to it to see what's up.