LazeMSS/OctoPrint-TopTemp #40: Pi throttled after uninstall

SingularityJ, 2021-03-04:
just uninstalled and now my pi is staying throttled so i cant re install, im guessing its not getting a temp value anymore so is throttled. im going to try some different front ends now anyway so no biggie but worth looking into

LazeMSS, 2021-03-04:
There is a know problem with octoprint - it don't allow you to install plugins if under powered - its not a problem in my plugin.

closed 2021-03-04 by LazeMSS

SingularityJ, 2021-03-04:
Fair enough, I'm trying something different for now anyway but I do have a
dedicated mean well 5v 4a suppy set at 5.1v and never had a problem before
so that's a little surprising Not looking to troubleshoot just thought I
would mention incase, great plugin

LazeMSS, 2021-03-04:
Okay