LazeMSS/OctoPrint-TopTemp #44: [FR] Add ability to update Part Cooling Fan regardless of how its speed was set

lightmaster, 2021-03-07:
If you enable `#define HOST_ACTION_COMMANDS` and `#define HOST_PROMPT_SUPPORT` within https://github.com/MarlinFirmware/Marlin/blob/b26516b3ac7cd7d15a751af4b299a38ac8ac7a57/Marlin/Configuration_adv.h#L3264-L3268, then Marlin will echo back over terminal a lot of commands, including the fan speed. This includes setting fan speed from the basic LCD with an encoder wheel and also TFTs touch screen. This should be able to greatly expand the available options of TopTemp and increase reliability of it being able to recognize when something changes.

LazeMSS, 2021-03-07:
Yes and if you do this then my plugin can filter for these results. I don't have a printer with marlin firmware so I can't test it. But if you have an example of what is returned when having this enable then maybe I can help but it's not something that needs to coded

lightmaster, 2021-03-07:
`M106 P0 S127` is what it sends to terminal

lightmaster, 2021-03-07:
I switched it from GCode Sent to From Printer and it works. Did not notice that option before.

lightmaster, 2021-03-07:
What about combining GCode Sent and From Printer such that if you do not have `HOST_ACTION_COMMANDS` enabled, it will look at the gcode that you sent, and if you do have it enabled, it can see both the gcode you send and the gcode the printer reports back.

LazeMSS, 2021-03-07:
Well I think people should setup what fits their printers.

LazeMSS, 2021-03-07:
... and not quite sure how you wanted that to work

lightmaster, 2021-03-07:
I was just thinking that they become a single combined entry. If it sees both a sent gcode and received from printer gcode, then they'd be the same since it's echoing back. Otherwise if it sees just a sent gcode, then it shows just the sent gcode.

LazeMSS, 2021-03-07:
Yes both that would require even more monitoring.
Could you provide an example of how this should work. I don't want to implement something where my plugin sends gcode/commands to the printer

lightmaster, 2021-03-17:
> If you enable `#define HOST_ACTION_COMMANDS` and `#define HOST_PROMPT_SUPPORT`

I was way off on the setting that I was talking about, its actually `#define REPORT_FAN_CHANGE`. And this only allows you to see the fan speed changes from any source, so probably not worth the effort to needlessly combine the Gcode Sent and From Printer bits into one.

LazeMSS, 2021-03-17:
So close?

closed 2021-03-17 by lightmaster