LazeMSS/OctoPrint-TopTemp #45: Shown timeline should be always the same

Fank, 2021-03-09:
Bed, Tool 0 and CPU temperature shows alway a max of 10 minutes:   
[image]

But Cooling fan speed shows a completly different timeline:   
[image]

It looks like it only counts the time when this option is enabled:    
[image]

LazeMSS, 2021-03-25:
I have a fix on the way.

But the cooling fan speed is only updated when a gcode is sent to alter the speed thats why its sporadic.

LazeMSS, 2021-03-25:
Fixed/closed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.3

Please reopen if not fixed.

closed 2021-03-25 by LazeMSS