LazeMSS/OctoPrint-TopTemp #47: Help for Fan in %

luco85, 2021-03-25:
Hello! i not understand how to have the fan value in %
So i add custom tab
- type: gcode send
- deactivate "input is a temperature value
- i use layer fan speed from menù
- ^M106.*?S([^ ]+) formula appears to sniff gcode

In post calculate i stop. i Try many value for calculation as "x" for x/255*100
How formula i need?
thanks

LazeMSS, 2021-03-25:
Uppercase X?

luco85, 2021-03-25:
Ya X sorry, i wrote wrong here
What formula i need to use?

LazeMSS, 2021-03-25:
`X/255*100`

My settings:
[image]

luco85, 2021-03-25:
Ok thanks! I’ll try

LazeMSS, 2021-03-26:
Okay closing. Reopen/comment if still not working

closed 2021-03-26 by LazeMSS

luco85, 2021-03-26:
It work thanks! I thought that X was a generic parameter for example