LazeMSS/OctoPrint-TopTemp #52: With no icon, Top Temp text does not align with other text (class navbar-text) on navbar

ManuelMcLure, 2021-08-24:
This is an example with Navbar Clock and Navbar Temp enabled, Discorded theme in UI Customizer:
[image]
If I select "Align icons to the left" it looks like:
[image]
Both Navbar Clock and Navbar Temp just set the class to "navbar-text"
With icons and "Align icons to the left":
[image]

LazeMSS, 2021-08-24:
Thanks - will be fixed soon.

LazeMSS, 2021-09-07:
Fixed in 0.0.1.5

closed 2021-09-07 by LazeMSS