LazeMSS/OctoPrint-TopTemp #55: read data from DisplayLayerProgress

electronicm, 2021-09-09:
You are a genius at UI customization!
I like the look and functionality of this plugin, as well as UI Customizer.

I would like to add the following values to the header with icons:
- currentLayer/totalLayer
 totalLayer
- printerState
- printTimeLeft
- estimatedEndTime
- 
These are already provided by the plugin DisplayLayerProgress. However, these would currently have to be fetched, for each value individually, in a very resource consuming way.
It would be a more user-friendly and resource-saving way if you could integrate the data, for example via EventHandler.

LazeMSS, 2021-09-09:
They are on the nice to list

puterboy, 2022-08-16:
You can easily get print time left (as well as remaining percent) using a custom parameter.

This is what I do for time remaining:
*Name: Time Remaining
Type: From printer
Regexp: ^NORMAL MODE:.*print time remaining in mins: (\d+);.*
(or if using silent mode: ^SILENT MODE:.*print time remaining in mins: (\d+);.*
Post calculate: X+1
    NOTE: Add '1' minutes since when says '0' really up to 1 minute left
+Only show while printer is printing
-Input is a temperature value
+Show overlay with more details on mouse hover
Unit: m
Label: "Time: "
Width: 0
Icon: fas fa-hourglass-end
-Colorize icons to indicate hot or not
Number of digits: 0
Decimal seperator: .
+Show graph backup
Graph transparency: 0.2
Graph height: 50
Graph line width: 1
Graph line color: <Black>

For percent completed, I do:

Name: Percent  Printed
Type: From printer
Regexp: ^NORMAL MODE: Percent done: (\d+);.*
(or for silent mode: ^SILENT MODE: Percent done: (\d+);.*
Post calculate: <blank>
+Only show while printer is printing
-Input is a temperature value
+Show overlay with more details on mouse hover
Unit: %
Label: "Done: "
 Width: 0
Icon: fas fa-percent
+Colorize icons to indicate hot or not
Color change level: 95
      NOTE: Change color when 95% completed
Number of digits: 0
Decimal seperator: .
+Show graph backup
Graph transparency: 0.2
Graph height: 50
Graph line width: 1

puterboy, 2022-08-16:
Note estimated end time can be calculated from remaining time and current time

LazeMSS, 2022-12-11:
Closing - this can be done as Puterboy wrote :)

closed 2022-12-11 by LazeMSS