LazeMSS/OctoPrint-TopTemp #56: [FR] show all passed decimal places

electronicm, 2021-09-10:
Is it possible to display all the decimal places passed without specifying a fixed length?
For example, if the field "Number of digits" is empty.

This would help me a lot, because I am currently still trying to map the layer state.
Your plugin checks if the return value of the RegEx is an int or float.
Therefore I have to send via 'DisplayLayerProgress' `Layer [current_layer].[total_layers]` to the printer.
Per 'Gcode sent' I then check for `^M117 Layer ([\d\.]+)` and output `/` as decimal character.
Since the maximum number of layers can range from one to six digits, the fixed length of the decimal places leads to the wrong display of the maximum layers.

Example1:
```
Number of digits: 4
Current layer = 5
Max Layers = 256
-> Output to printer: M117 Layer 5.256
-> RegEx match: 5.256
-> Output to TopTemp: 5/2560
```

Example2:
```
Number of digits: 2
Current layer = 5
Max Layers = 256
-> Output to printer: M117 Layer 5.256
-> RegEx match: 5.256
-> Output in TopTemp: 5/25
```
The expected output of both examples would be `5/256`.

LazeMSS, 2021-09-11:
Not sure if I understand what you mean - can you explain a bit more.

I will probally make the following new "inputs" in the next version:
- layer height/z
- progress in %/minutes
- remaining in %/minutes
and more

electronicm, 2021-09-11:
At the moment:
If I currently enter 2 in 'Number of digits' and
- receive '114.3418' via RegEx, '114.34' is output
- recive '5.27' via RegEx, '5.27' is output
- recive '7.1' via RegEx, '7.10' is output

If I currently enter 0 in 'Number of digits' and
- receive '114.3418' via RegEx, '114' is output
- recive '5.27' via RegEx, '5' is output
- recive '7.1' via RegEx, '7' is output

However, I need the possibility that the decimal places are not adjusted.
If I leave the 'Number of digits' field blank or whatever you define and
-recive '3.243' via RegEx, '3.243' is also output
-recive '5.27' via RegEx, '5.27' is also output
-recive '7.1' via RegEx, '7.1' is also output
The output of the RegEx should be passed on in the code without adjusting the places after the comma.

Afterwards, it should continue to genes normally in the code and replace the '.' with the character defined under 'Decimal separator'.

LazeMSS, 2021-09-11:
Ahh NOW i get it - will fix it right away

electronicm, 2021-09-11:
> Not sure if I understand what you mean - can you explain a bit more.
> 
> I will probally make the following new "inputs" in the next version:
> - layer height/z
> - progress in %/minutes
> - remaining in %/minutes
> and more

Do you also have something like
- [current layer] / [max. layer] -> 5/234
- ETA
And can you also take into account the values ​​of PrintTimeGenius?

LazeMSS, 2021-09-11:
Yes think so.

LazeMSS, 2021-09-20:
Included in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.6

closed 2021-09-20 by LazeMSS

electronicm, 2021-09-25:
It works great after I figure out what I need to enter.
The help text in d7e507e toptemp_settings.jinja2 line163 is only displayed in the bed settings.

LazeMSS, 2021-09-25:
Thanks will fix ASAP