LazeMSS/OctoPrint-TopTemp #57: [FR] Spacer

electronicm, 2021-09-10:
It would increase clarity if you could place separators between icons.
Not so Importent: It would be even more ingenious if you could bind these separators to the status "Only show while printer is printing", "Hide when not operational", "Only show while icons left/right are displayed".

Example idle:
CPUTemp   CPULoad

Example heating bed:
Bed   |   CPUTemp   CPULoad

Example printing:
Layer   %Completed   |   Bed   Tool0   |   CPUTemp   CPULoad

LazeMSS, 2021-09-11:
Working on it.

LazeMSS, 2021-09-20:
Included in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.6

closed 2021-09-20 by LazeMSS

electronicm, 2021-09-25:
This works also like a charm.
Is there still a possibility that you add the option "Show separator on left" or a DropDown with "none, left, right, both"?

LazeMSS, 2021-09-27:
Just to understand - why do you want the seperator on the left etc. I understand the grouping of items but why left, none etc.