LazeMSS/OctoPrint-TopTemp #59: After Update Numbers Don't line up

ryancasler, 2021-09-29:
After Updating to the latest version, the number appear at the very very top of the window, almost off the top of the screen.
[image]

LazeMSS, 2021-09-29:
Can you post a screenshot of your settings?

LazeMSS, 2021-10-01:
@ryancasler i need to know a bit more about your configuration.

LazeMSS, 2021-10-02:
@ryancasler  ping....

ryancasler, 2021-10-03:
Yes, here are my settings:
[image]
[image]
I didn't see anything about the spacing above the numbers, so I wasn't really sure what my settings would tell you.

LazeMSS, 2021-10-03:
Will check asap

ryancasler, 2021-10-05:
I don't have UI customizer but I do have a bunch of other plugins running.  Tasmota, TouchUI (but not using the TouchUI on this device), Floating navbar (tried disabling this one and it didn't help) and a few others, none of which seem to have any UI consequences.  If there are any logs or anything you need me to grab, just let me know.

LazeMSS, 2021-10-05:
Great I will look today

ryancasler, 2021-10-30:
Yup....looks great!  Thanks so much!!