LazeMSS/OctoPrint-TopTemp #66: octopi bullseye CPU temp `vcgencmd` path changed

idcrook, 2021-12-02:
Out of the box "CPU Temperature" Test:

```shell
 /opt/vc/bin/vcgencmd: not found
```

Logging into shell:

```shellsession
$ command -v vcgencmd
/usr/bin/vcgencmd
```

OK, so in Raspberry Pi OS/bullseye, the path to this command has changed.  

## Functioning Replacement

Replacing the command path in the "CPU temperature main settings" panel, it works as expected. 

full command (_just the path changed_):

```shell
/usr/bin/vcgencmd measure_temp|cut -d "=" -f2|cut -d"'" -f1
```

LazeMSS, 2021-12-02:
It should be supported:
https://github.com/LazeMSS/OctoPrint-TopTemp/blob/0e1ef0ce612624d2268f1e0d24b01274bb5ca4c3/octoprint_toptemp/__init__.py#L430

idcrook, 2021-12-02:
This was on a bullseye-based octopi build "out of the box", which can be found in its nightly builds.  Provided as a "heads-up", as it does not seem to auto-recognize correctly on this upcoming octopi image.

Reiterating: upon plugin installation, the incorrect path/missing command was what the "CPU Temperature" option form had in it. Manually changed the command path in the form, it then worked.

LazeMSS, 2021-12-04:
Thanks I will look into it ASAP

LazeMSS, 2022-01-02:
Could you trying using the develop branch:
https://github.com/LazeMSS/OctoPrint-TopTemp/tree/develop

What happens if you run the two commands from a normal bash?

idcrook, 2022-01-04:
> Could you trying using the develop branch: https://github.com/LazeMSS/OctoPrint-TopTemp/tree/develop

When I next power on the printer. Would this be?

 1. uninstall plugin 
 2. ?? any cleanup ??
 3. install plugin manually from that `develop` branch?

> What happens if you run the two commands from a normal bash?

the error message (or successful) as described in this issue opening.  

It was not a big issue to fix - plugin was just not properly auto-detecting the existing path, and was straightforward to manually specify the path existing on the system

> And did you do a import old settings from a backup?

No import.  Install from octoprint plugins interface.   was on a bullseye-based octopi build "out of the box" from <http://unofficialpi.org/Distros/OctoPi/nightly/?C=M;O=D>

LazeMSS, 2022-01-05:
Yes uninstall (will do a cleanup) and the install from the develop branch. I have not changed that much I tried change the command names to foobar and it worked.

closed 2022-12-03 by LazeMSS