LazeMSS/OctoPrint-TopTemp #67: Feature request: Cooling Fan speed before pause, exposed as variable.

daimon82, 2021-12-24:
Scenario:
When pausing a print in octoprint one needs to manualy set gcode in sections: Gcode after pause and gcode before resume, 
to retract filament, move printhead, set the nozzle temperature, set the heater timeout (prusa) and set the cooling fan.

Problem:
There is no variable in octoprint that stores last value of the cooling fan, and so it's impossible to restore the correct value when resuming the print.

Resolution:
Your plugin could keep two values for cooling_fan: actual and before print is paused.
You can add a script hook to your plugin and expose a dictionary variable for specified gcode section:
https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-scripts

 `	def gcode_script_variables(self, comm, script_type, script_name, *args, **kwargs):
		if not script_type == "gcode" or not script_name == "beforePrintResumed":
			return None
        
		prefix = None
		postfix = None
		variables = cooling_fun_speed           #dictionary type object containing 'coolingfan_before_pause' value
		return prefix, postfix, variables

__plugin_name__ = "Top Temp"
__plugin_pythoncompat__ = ">=2.7,<4"

def __plugin_load__():
    global __plugin_implementation__
    __plugin_implementation__ = TopTempPlugin()

    global __plugin_hooks__
    __plugin_hooks__ = {
        "octoprint.comm.protocol.scripts": __plugin_implementation__.gcode_script_variables
    }`

Then it will allow to use Following "gcode before resume script" to restore :
M106 S{{ plugins.toptemp.coolingfan_before_pause|int() }}

I tried to make those changes myself but I'm too green when it comes to python and octoprint. 
You would make your plugin almost a must have for people who want to pause their prints overnight.
I hope my suggestion is helpful, merry xmass :)

daimon82, 2021-12-27:
I found that M601 and M602 commands will take care of pausing and unpausing a Prusa printer, theres no need for extensive gcode scripting and plugins. Case closed.

LazeMSS, 2021-12-28:
😁

closed 2021-12-28 by LazeMSS