LazeMSS/OctoPrint-TopTemp #70: Help with regexp for a special temp sensor needed

Caipitrooper, 2022-04-03:
Hi,
like issue #18 I too have the problem of getting the temperature back from a temperature sensor that is
replied from a M105. My temperature sensor is named "M:". I set it up this way in Klipper and this is the temp sensor
of the STM32 on the Spider board.

The M105 looks like this:

[image]

I set the custom temp sensor and regexp to this:

[image]

Regarding to the Regex Tester (https://de.infobyip.com/regularexpressioncalculator.php) it should give back the right value:

[image]

However, I just get a waiting circle:

[image]

What would be the correct expression to get the temperature displayed?
Thanks a lot in advance.

LazeMSS, 2022-04-04:
Maybe just try:
M:([^ ]+)

https://regex101.com/ is a very good regexp editor

Caipitrooper, 2022-04-05:
It worked with this regexp, thank you very much. Issue can be closed.

closed 2022-04-05 by Caipitrooper