LazeMSS/OctoPrint-TopTemp #72: FR: add option to show, if temp is over XXX

electronicm, 2022-06-15:
It would be nice, when some values has additional display-trigger.
For example, only show if
- bed is over XXX °C
- nozzle is over XXX °C
- CPU is over XXX °C
- Usage is over XXX %

On some material - bed combinations, you need a specific bed temperatur for easy removing.
With this lazy-man-feature, i can easy watch at the head of Octoprint if it is time to go to the printer, without showing the temperaure all the time.

LazeMSS, 2022-06-16:
You could use the "Color change level"?

electronicm, 2022-06-16:
> You could use the "Color change level"?

I am currently already using the mentioned option. However, I had hoped that with the new option I would not have to permanently display the temperatures. Freely according to the motto: Show only what is currently necessary.

Also, the option would be an easy and safe way to "capture" certain operating conditions faster.
For example:
- Do I see an icon for Nozzle or Bed => Needs more time to cool down.
- Do I see the symbol for CPU Temp? => Must pay attention to the board fan.
- Do I see the symbol for CPU Load? => Must check the processes.
- Is a symbol red? => (Injury) danger!
- Is the bed still yellow? => I still have time for a coffee.

Specifically for nozzle and bed I would aim for the following display behavior:
- Printer is idle and everything cold-> no symbol to be seen.
- Temperature set (printing process/leveling) -> icon is displayed.
- Printer idle, but temperature is still above a certain temperature -> icon is displayed.

My desired (three) color behavior:
- Bed too warm for easy removal (> 40 °C) -> color: yellow
- Nozzle too warm for changing (> 35 °C) -> color: yellow
- bed/nozzle above touch temperature (>60 °C) -> color: red
- everything else in default color

Now I have thought a bit further than actually intended.
What do you think about the whole issue?

Translated with www.DeepL.com/Translator (free version)

closed 2025-06-22 by electronicm