LazeMSS/OctoPrint-TopTemp #73: Struggling to remove "custom" fields

bladeslapper, 2022-06-19:
I have 2 custom items in my top temps setup (to monitor the Pi itself) that I was using to fault check. I'd like to remove them but they keep re-appearing after I "delete" and then "save". I've managed to re-name them both to "delete" and "delete2" but they won't go away.
Running octoprint 1.8.1Python 3.7.3 OctoPi 0.18.0

LazeMSS, 2022-06-19:
Strange - what happens if you delete one of them and then click save?

bladeslapper, 2022-06-19:
> Strange - what happens if you delete one of them and then click save?

It just stays there looking smugly at me! It lets me edit it no problem but not get rid completely. I even managed to delete all the data so it just shows the loading circle but can't make it actually disappear.

der-pw, 2022-07-07:
Same Problem here.
It's not possible to delete the items. I click on delete and after that on save, but the filed isn't deleted after that.

LazeMSS, 2022-07-13:
Found the bug - it works if you restart octoprint after saving settings - im trying to figure out why the settings are not reloaded

bladeslapper, 2022-07-29:
> Found the bug - it works if you restart octoprint after saving settings - im trying to figure out why the settings are not reloaded

Not for me. I can delete these over and over and they just hang around despite restarts.

meier2009, 2022-12-03:
no Update?

tried everything, but also still have these anoying circles and data i dont want...

LazeMSS, 2022-12-03:
Sorry Will look later today

meier2009, 2022-12-03:
No Hurry :) 
Was just a quaestion! its a nice tool, by my bar is always full, so the extra icon dont help, hehe

As i tried something else because recognized strange behavior. It seem that only ONE Change/Delete is made, if you try a second, the first will be recoverd.

i deleted all data in pluginmanager, just to only have 2 and my CPU Temp item and not the other 10 ;)

and then setup only cpu temp and deletet fan (becaus I dont ned) and it was gone.
As tried to delete cpu also, fan recoverd and vice versa.

also, maybe add the funktion to HIDE it, like in Tool and Bed, in every tab, will help faster, than try figure out why it dont delete ;)

LazeMSS, 2022-12-11:
Should be fixed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.1.8 - please comment it/reopen it has not fixed it.

closed 2022-12-11 by LazeMSS

meier2009, 2022-12-13:
hi,
cant test 0.0.1.8, maybe this was okay for me.

now in version 0.0.2.0 cant make user added , it doesn´t do anything.
nice to have 10 extra tool settings, but can´t do for example cpu temp now

maybe you can explain how to install 1.8. to test if this is enough for me!? 
:)