LazeMSS/OctoPrint-TopTemp #75: Can't get Background graph line color to change

puterboy, 2022-08-11:
When I click on it, I can change the colors in the popup but they don't "stick"

LazeMSS, 2022-12-03:
Are you using any themes etc?

LazeMSS, 2022-12-11:
This bug is due to themes overwriting it - I will make an option to force the settings

LazeMSS, 2022-12-11:
Should be fixed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.0 - please reopen if not working/fixed as expected

puterboy, 2022-12-21:
Unfortunately, still doesn't seem to be working for me

LazeMSS, 2022-12-21:
Yes there is a small UI bug in settings with showing the colors in the selector. But the color change is stored and updated in the UI.

Are you using any themes? Since you say the colors are blue?

And what color of the graph do you expect it to change?

puterboy, 2022-12-22:
I use Themeify to set the color of Toptemp to blue (and the weight of the text to bold) via the following line:

```
Selector                                           CSS-Rule                 Value
 #navbar_plugin_toptemp                  font-weight             bold
 #navbar_plugin_toptemp                  color                       #0000ff
```
I was hoping that I could change the color of the lines in both the overlay icon graph and in the larger hover popup graph. Most importantly, I was hoping to change the color of the overlay icon graphs so that the color would contrast better with the (blue) text and icon colors set by the theme.

LazeMSS, 2022-12-22:
Did you check "Overrule themeing" :) checkbox

you could also consider using ".TopTempText" iinstead of "#navbar_plugin_toptemp"

LazeMSS, 2022-12-22:
Should be fixed now: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.1

closed 2022-12-22 by LazeMSS

puterboy, 2022-12-22:
Thanks! Works!!