LazeMSS/OctoPrint-TopTemp #78: [Feature Request] Ability to trigger action if temperature too low/too high

puterboy, 2022-08-11:
Action could be either a user-specified system command or Gcode to send to printer if temperature is too high or too low based on the colorization cutoffs (or maybe a separate user-specified number).

I know this is not directly related to displaying the temps but since your *excellent* plugin collects all this data in one place, it would be an easy and natural place to trigger an action...

LazeMSS, 2022-08-12:
Out of scope - I think there are other plugins that will handle this.

closed 2022-08-12 by LazeMSS

puterboy, 2022-12-21:
I understand that it could be out of scope but it seems a shame to have repeat the data collection with another plugin. So still would be nice :)