LazeMSS/OctoPrint-TopTemp #79: [BUG] Can't change sort order since conflicts with Upload locally/Upload to SD card

puterboy, 2022-08-11:
When I try to change the sort order on the "General" tab by using my mouse to move the items, the screen darkens and gives me the option on the left to "Upload locally" and on the right to "Upload to SD card".

To change the sort order, I had to manually move lines in config.yaml as a workaround.

LazeMSS, 2022-08-12:
Hmm... im running UI Customizer - it handles this bug so maybe try installing that?

puterboy, 2022-08-12:
I have been  using Themeify...
But as I said, I solved it for myself by manually editing config.yaml but others might not be comfortable with that :)

LazeMSS, 2022-08-13:
You should really use UI customizer. Much better 🤪

I will implement the drag and drop fix ASAP

puterboy, 2022-08-14:
Thanks.
I'm sure it's much better :) just haven't had the time yet to play with it :)

LazeMSS, 2022-12-11:
Should be fixed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.0 - please reopen if not working/fixed as expected

closed 2022-12-11 by LazeMSS

puterboy, 2022-12-21:
Thanks! Works great!

puterboy, 2022-12-21:
Thanks! Works great now!