LazeMSS/OctoPrint-TopTemp #80: [Feature Request] Add  option for sensors to "Only show when printer connected"

puterboy, 2022-08-11:
Sorry for all these wish list requests but I really love your plugin :)

Some custom options like the Einsy board temperature are only available when the printer is connected (not just when printing) -- so if not connected, then I get a "spinning thing" in the toolbar.

LazeMSS, 2022-08-12:
Will look.

LazeMSS, 2022-12-11:
Implemented as "Hide if printer is disconnected" is next version

LazeMSS, 2022-12-11:
Should be fixed in https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.0 - please reopen if not working/fixed as expected

LazeMSS, 2022-12-21:
Sorry this does not make sense to me...

1.  Could you post screenshot of the temp sensor you are trying to hide and maybe explain in more details what your expecting

2. if you disconnect the printer while octoprint is running then yes the printer is offline

puterboy, 2022-12-22:
The only reason I had requested the "Hide if No Printer" feature was to cover the case of the Einsy temperature sensor but now I realize that this feature request may not even be necessary since ideally the whole "From printer" type should only display if the printer is connected.

Also looking at the code in TopTemp.js, it seems like the sensor should not be displayed if there is no valid data regardless of the value of hideIfNoPrinter
```
            // Do know this or want it shown
            if (typeof iSettings == "undefined"
                || iSettings.show() == false
                || data.actual == null
                || data.actual == undefined
                || (data.target == 0 && iSettings.hideOnNoTarget())
                || (!customType && self.settings.hideInactiveTemps() && self.tempModel.isOperational() !== true)
                || ('waitForPrint' in iSettings && iSettings.waitForPrint() && !self.connection.isPrinting())
                ||  ('hideIfNoPrinter' in iSettings && iSettings.hideIfNoPrinter() && !self.tempModel.isOperational())
            ){
                $('#navbar_plugin_toptemp_'+name).hide();
                $('#navbar_plugin_toptemp_'+name+'_divider').hide();
                return;
            }else{
                $('#navbar_plugin_toptemp_'+name).show();
                $('#navbar_plugin_toptemp_'+name+'_divider').show();
            }
```

I haven't dived deeply into the code, but shouldn't "data.actual == null" and/or "data.actual == undefined" alone be sufficient to hide the sensor when the Printer is not connected? And if true, the "hideIfNoPrinter" functionality may not even be necessary in the first place.

Either way, I'm not sure why the Einsy sensor displays when the printer is off...

Anyway, attached is a screenshot of what happens when I restart Octoprint with the printer off (and hence disconnected):

Note that the first whirling-circle is supposed to be the Outdoor temperature but it is now about -3degC outside (i.e. cold!), so that fails as per my other bug report. The last whirling circle is the one caused by the printer being off when trying to get the Einsy temperature (as per the config above).

[image]

LazeMSS, 2022-12-22:
Basically the logic is using the standard octoprint way of handling stuff - based on your screenshot and if you have checked
"Hide if printer is disconnected" then the last spinning circle should not be there but maybe its because it delayed/not getting data due to the other bug.

LazeMSS, 2022-12-22:
Should be fixed now: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.1

closed 2022-12-22 by LazeMSS

puterboy, 2022-12-22:
Thanks! Works!!

puterboy, 2022-12-26:
Minor display bug here. When the sensor first turns on, there is no *leading* white space around the spacer. If printer is turned off and then on again, the white space appears correctly.

LazeMSS, 2022-12-27:
> Minor display bug here. When the sensor first turns on, there is no _leading_ white space around the spacer. If printer is turned off and then on again, the white space appears correctly.

Please make new issues instead of adding to existing issues when not related. The missing spacing is probably missing due the way the spacing is made by giving the last item a margin of 0px - but im not 100% sure - i could not recreate - it could be due to the number of sensors shown/hidden etc.