LazeMSS/OctoPrint-TopTemp #81: [Feature Request] Add ability to display arbitrary text string (and not just a number)

puterboy, 2022-08-11:
For example a status or a time.
If so, then the graph options would be disabled

LazeMSS, 2022-08-12:
Where? How - could you do a mockup

puterboy, 2022-08-12:
I would add a new Custom "Type", say called "String" with the following relevant subset of config options for the 'Command' type
```
Name:
Type: String
Command: <Enter command that generates desired string for display>
Run interval: 
Only show when printer is running
Label:
Width:
Icon:
```

puterboy, 2022-08-14:
Actually, on second thought, I think a simpler/cleaner solution would be to use the existing Type=command with the following modification:

```
Name:
Type: Command
Command:  <Enter command that generates desired string for display>
Run interval: 
Only show when printer is running:
Input is:
     -  A temperature value
     -  Number
     +  String [Select string]
Label:
Width:
Icon:
```
When string is selected all the other options are hidden.