LazeMSS/OctoPrint-TopTemp #87: [BUG] Custom "Command" type doesn't allow negative values

puterboy, 2022-12-21:
I am using TopTemp to display the outside temperature which in the Northeast U.S. is now negative in Celsius.
This causes an error since it seems that command is expected to be a positive number.
There are other sensors which could presumably give a negative number too.

LazeMSS, 2022-12-22:
Should be fixed now: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.1

puterboy, 2022-12-22:
I think there is still a bug here. The 'test' now works (per your code fix) but it still won't *display* a negative number -- it just gives a spinning wheel...

puterboy, 2022-12-23:
OK - I found the problem.
Try this fix:

```--- __init__.py.~1~     2022-12-22 09:17:54.000000000 -0500
+++ __init__.py 2022-12-23 13:42:35.678604481 -0500
@@ -729,7 +729,7 @@
     def handleCustomData(self,indx,out,time):
         self.debugOut("Got custom data: " + str(out))
         # Check
-        if isinstance(out,(float, int)) or str(out).replace('.','',1).isdigit():
+        if isinstance(out,(float, int)) or str(out).lstrip('-').replace('.','',1).isdigit():
             resultData = [time,float(out)]
             if indx not in self.customHistory:
                 self.customHistory[indx] = []
```

Only limitation here is that '--2.3' (or any number of initial dashes) would be recognized as a number.
If you want to eliminate that case you would need to either add more logic or use 're' for regex test.

closed 2022-12-27 by LazeMSS