LazeMSS/OctoPrint-TopTemp #92: Custom Temp Not Updated

ryancasler, 2022-12-31:
I have a custom temp that updates by reading a txt file on my Pi.  I put the command into the plugin and the test works correctly and reads the correct temp.   But the temp displayed on the menu bar is incorrect.
[image]
[image]
[image]

LazeMSS, 2022-12-31:
Strange - my first thought was maybe some fahrenheit conversion but that does not match the numbers you posted.
Tried on my own setup and it works fine.

ryancasler, 2023-01-04:
And it does subsequently on my system as well.  Must have been a random failure.  I will continue to monitor and re-open the issue if it pops up again.

closed 2023-01-04 by ryancasler