LazeMSS/OctoPrint-TopTemp #93: BUG: Sensor error/offline requires restart of octoprint

puterboy, 2023-01-08:
I had a sensor offline for a while so that the custom command returned some non-numerical string.
When the sensor came back online, it still had the display frozen at the last good value and the graph showing no data with the x-axis labeled as all zeros and the avg/min/max showing Null.

Refreshing the web browser didn't help.
Going into the settings for TopTemp and running 'test' showed that the sensor command was working again... just somehow the lack of good data had frozen the display and borked the graph history.

The only way I could fix it was by restarting octoprint which you obviously can't always do - for example if you are in the middle of a print.

The lack of sensor data (or even the presentation of 'bad' sensor data) should be handled gracefully... :)

puterboy, 2023-01-08:
Actually, it seems like even one bad sample can trigger this...

puterboy, 2023-01-08:
Note my ideal behavior in case of invalid or absent sensor data would be the following:
1. Sensor displays NA in the ribbon
2. Histogram skips the data point

LazeMSS, 2023-01-08:
This I can not recreate - tried using a file for input and deleting it - basically it just skips the data
Also tried creating a file returning junk - no problems
Also tried creating a command that would cause a timeout.

Could you explain in details how to recreate - what data does the sensor return when failing?

puterboy, 2023-01-08:
It seems to break if the command returns nothing.

Until that is fixed, try this to replicate:

bash -c "if [ -e /tmp/crap ]; then /bin/true; else date +%M; fi"

It should give a graph cycling from 0 to 60 (corresponding to the time in minutes)

Then after a few samples, type "touch /tmp/crap" in a shell (which simulates the sensor returning nothing)
The displayed data is frozen at the last valid minute (as expected perhaps, though I would argue, the display should be NA).
And then the histogram goes to all zeros labels on the X-axis and avg/min/max all equal to Null which.

Then after a few minutes type "rm /tmp/crap" in a shell (which will simulate sensor returning to normal).
However, the above error state still persists!
Even worse, it seems to require a restart of Octoprint to fix.

I believe this is related to my latest comment re-opening https://github.com/LazeMSS/OctoPrint-TopTemp/issues/88
i.e., having a command that returns nothing (e.g., /bin/true) causes the sampling to hang forever.

puterboy, 2023-01-15:
Were you able to replicate with the above explanation?

LazeMSS, 2023-01-15:
Have not tried yet.

LazeMSS, 2023-01-15:
Fixed in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.3

puterboy, 2023-01-20:
I am still getting the problem whereby if a sensor is offline (and thereby returns an error) that the graph fails even when the sensor resumes... And that the problem can only be reset by restarting Octoprint.

i.e., still get a blank graph with NA for avg/min/max..

The correct behavior would be just to "skip" the time series points without valid data -- i.e. don't display the value (just display NA) or the graph (blank or gap in the graph) for those points and don't include them in the min/max/average.

puterboy, 2023-01-30:
@LazeMSS - any ideas why this is still occuring?

LazeMSS, 2023-01-30:
I'm on holiday will look later

puterboy, 2023-01-30:
Enjoy your vacation!!!! I'm sure it's well-deserved.
LazeMSS wrote at about 21:06:14 -0800 on Sunday, January 29, 2023:
 > I'm on holiday will look later 
 > 
 > -- 
 > https://github.com/LazeMSS/OctoPrint-TopTemp/issues/93#issuecomment-1408004528
 >

LazeMSS, 2023-02-03:
I still can't recreate this.

puterboy, 2023-05-15:
Trying to figure out how to recreate this reliably.
This happens with only of my 'sensors' that uses 'rclone' to read a value on a cloud-drive. 
After a seemingly random amount of time the sensor stops updating the data and the value displayed in the NavBar itself is stuck at the last valid before that point.
The histogram itself is either stuck at that same point or it displays "NaN" and a flat line.

The only thing I can think of is that reading the cloud value times out sometimes and that causes the sensor to stop recording -- or maybe it is still waiting for the sensor read to return and hence never gives up and moves on.

I know based on an earlier bug report, you added a timeout to the 'test' function... is there also a timeout for the regular sensor display intervals -- i.e., if the sensor value fails to return within N seconds, it stops trying for that period and tries again (from scratch) the next period?

puterboy, 2023-05-15:
Actually, fixing the problem in #88 doesn't fix the problem,
Specifically, the expression works under 'test' but when running normally it runs until TSTAR but then doesn't come back after TEND.

puterboy, 2023-07-17:
Anything I can help do to fix the problem?