LazeMSS/OctoPrint-TopTemp #94: BUG: Histogram length doesn't work for faster sampling rates

puterboy, 2023-01-08:
I set the "Show data this many seconds old on graph" to 3600 seconds (60 minutes) for all my sensors.
For the sensors that update every 5 seconds, only about 26 minutes of data is displayed.
For the sensors that update every 30 seconds, the full 60 minutes of data is shown.

I suspect that not enough buffer is reserved to hold the number of samples required to sample 5x/sec times 60 minutes.
Presumably enough buffer data should be allocated or alternatively, the UI should give a warning if the sampling rate times the histogram length is too large.

LazeMSS, 2023-01-08:
Fixed in the next release - please consider pick up coding and help fixing these bug by making pull request :)

puterboy, 2023-01-08:
Sure. I actually looked through the code but couldn't find the issue.

puterboy, 2023-01-15:
Any idea on when this may be released? Thanks

LazeMSS, 2023-01-15:
Yes hopefully tonight.

LazeMSS, 2023-01-15:
Fixed in: [BUG/FEATURE?] "Command" should have a timeout associated with it...

closed 2023-01-15 by LazeMSS