LazeMSS/OctoPrint-TopTemp #95: [Feature Request]

puterboy, 2023-01-08:
It would be helpful/convenient if custom commands were executed in the Bash rather than Bourne shell.
For example, 'echo' in the Bourne shell doesn't recognize the '-e' flag to allow for control characters like linefeeds.
This caused (unexpected) errors for me.
Of course it can be fixed by starting the command with "bash -c" but that is a bit messy.

Also at least in my default RPI setup, running Python and executing a shell command uses the Bash shell so the difference is unexpected.
-

LazeMSS, 2023-01-15:
Added in: https://github.com/LazeMSS/OctoPrint-TopTemp/releases/tag/0.0.2.3

closed 2023-01-15 by LazeMSS

puterboy, 2023-03-16:
Your fix seems to invoke 'bash' when running the command under 'test' but it doesn't seem to be running under bash when the actual command is run by TopTemp every N seconds.

I imagine that this should be a simple fix...

LazeMSS, 2023-03-16:
I would think it does - it runs using https://github.com/LazeMSS/OctoPrint-TopTemp/blob/7f5fc618b95fdf0facc25b0684c8f6bc7e7666e2/octoprint_toptemp/__init__.py#L574
which again calls
https://github.com/LazeMSS/OctoPrint-TopTemp/blob/7f5fc618b95fdf0facc25b0684c8f6bc7e7666e2/octoprint_toptemp/__init__.py#L876

The test and the timer uses the same.

puterboy, 2023-03-22:
Strange... I am having difficulty tracking down the error.
One interesting thing is that the timer version seems to run once correctly, then it fails to update subsequently on subsequent loops, leaving the initial value frozen in the navbar and the pop-up graph blank.
Of course, if I manually include 'bash -c' all works fine...