LazeMSS/OctoPrint-TopTemp #97: Graph doesn't display properly for the CPU fan predefined sensor

puterboy, 2023-01-16:
First, thanks for all the great progress in stamping out bugs!!!
Second, I noticed that the graph defined by the CPU sensor predefined value is not right:

1. The graph show a ramp up while the true graph is zig-zaggy (as confirmed by the mini-graph overlay and by the actual gcode)
2. The X-axis graph time is <60min even though it is set for 3600 seconds and has been printing for 90 minutes

[image]

Note that my configs for the sensor are as follows:
```
cu1:
        cmd: ^M106.*?S([^ ]+)
        colorChangeLevel: '1'
        colorIcons: true
        decSep: ','
        gHisSecs: '3600'
        gMax: ''
        gMin: ''
        graphSettings:
          color: '#af0096'
          forceStyle: true
          height: 50
          opa: '.5'
          show: true
          width: 1
        hideIfNoPrinter: false
        hideOnNoTarget: false
        icon: fas fa-fan
        interval: 25
        isTemp: false
        label: 'Fan '
        name: Cooling fan speed
        noDigits: 0
        postCalc: X/255*100
        show: true
        showPopover: true
        showSep: false
        showUnit: false
        type: gcOut
        unit: '%'
        updated: 1673843347689
        waitForPrint: true
        width: 0
```

Note that all my other sensors seem to be working fine whether built-ins whether custom shell commands or whether based on custom gcode regex.

Sorry for all the issues but hopefully this will help make your AWESOME plugin even better! Thanks!

puterboy, 2023-01-16:
And after 2 hours, the graph now shows a total error screen...

[image]

puterboy, 2023-01-16:
And then it resets and starts to make the triangle ramp pattern again... etc.
[image]

puterboy, 2023-01-16:
Note that the M106 messages in my gcode all seem well formed:
```
# cat file.gcode | grep -E "^M106.*?S([^ ]+)" 
M106 S76.5
M106 S76.5
M106 S76.5
M106 S76.5
M106 S51
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
M106 S51
M106 S76.5
...
```

LazeMSS, 2023-01-16:
I'm a bit confused you wrote CPU sensor, the graphs shows cooling fan speed.
So the problem is the graphs for the cooling fan speed?

puterboy, 2023-01-16:
Sorry for confusion. I meant Fan Speed.

LazeMSS, 2023-01-17:
Basically its because when the fan gets to a the final speed no more M106 commands are sent at that basically makes the data old. I will investigate a bit

puterboy, 2023-02-03:
The behavior here seems similar to what happens when a sensor returns no data https://github.com/LazeMSS/OctoPrint-TopTemp/issues/93

LazeMSS, 2023-02-03:
No its not the same behavior - what is happening here is that you want a graph that is 24 hours long on the x-axis but there is only data for the two first minutes. There is no easy fix for this

puterboy, 2023-02-03:
Couldn't you do something like if no new data, then assume fan speed unchanged and write the old fan speed as the current fan speed?
Indeed one might want to offer this option for any sensor as it might solve my other problem -- i.e., if no data or no valid data, then option to: 

1. Display last good value
2. Zero
3. Blank/no data - which I assume is the current case