LazeMSS/OctoPrint-TopTemp #98: Enhancement: Option to display X-axis times in hours...

puterboy, 2023-01-16:
For some of my sensors, I am looking at 24 hour time periods which is 86400 seconds -- leading to very crowded and difficult to read X-axis labels. Especially since they are sometimes written with a decimal point and digit after the decimal.

It would be great if either automatically or manually, one could switch from minutes to hours.
If done automatically, perhaps after a few hundred minutes, it would switch to hours.

Also, more generally, for minutes > 10 (and certainly > 100), I don't see any reason for a decimal point. Do you really need 3 digits of precision on the X-axis?

Thanks

puterboy, 2023-01-30:
In truth, the axis becomes impossible to read :)

LazeMSS, 2023-02-03:
Yes need to make it autoscale pretty

puterboy, 2023-05-15:
If full autoscaling is too complicated to implement at this time, perhaps at least drop the decimal point as that is probably hardly ever needed.