LazeMSS/OctoPrint-TopTemp #99: Run python script as custom command?

beffjessen, 2023-02-15:
I have a DHT22 sensor to monitor temperature and humidity in my printer enclosure. I had been using the Enclosure Plugin, but now that it's deprecated, there's no obvious alternative plugin... except maybe the custom command functionality in TopTemp?

I wrote a simple python script to grab the sensor temperature (based on [this code](https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/main/examples/dht_simpletest.py)) saved locally via nano, and it works when executed via SSH; however, I get a 'No such file or directory' error when I test `python3 example.py` as a custom command. Is it possible to execute python scripts as a custom command, and I'm just making a simple mistake? Or would this actually require a new feature be developed? I appreciate any help, thank you!

LazeMSS, 2023-02-15:
You need to provide the full path etc.

beffjessen, 2023-02-15:
Simple mistake it was! 😅

This is a great workaround for those of us who had been using the Enclosure plug-in for simple data output, hopefully others are aware of this as well. Thank you!

closed 2023-02-15 by beffjessen